mne-tools/mne-python #4304: Add exclude_bads argument to raw.get_data

cbrnr, 2017-06-06:
Sometimes it is necessary to explicitly get the raw data without bad channels. Currently, the amount of noise to achieve this is pretty high:

    raw.get_data(picks=mne.pick_types(raw.info, eeg=True))

Would it make sense to add a new argument `exclude_bads=False` to simplify this to:

    raw.get_data(exclude_bads=True)

?

agramfort, 2017-06-06:
why not just do

picks = mne.pick_types(raw.info, ...)
data, times = raw[picks, :]

I'd prefer that there is one recommended way to do things

cbrnr, 2017-06-06:
Hm, OK, but it's one extra line of code. I thought that since there already is a `get_data` method where you can make some selections on which data portions to get, maybe an extra argument to get only good channels made sense.

agramfort, 2017-06-06:
I think you can afford one extra line of code to keep the API simpler :)

cbrnr, 2017-06-06:
I agree, but excluding bad channels seems overly complicated on the other hand. And MEG-centric because only `meg=True` by default, so in reality this code would look like

    picks = mne.pick_types(raw.info, meg=False, eeg=True)
    data, times = raw[picks, :]

Here, `get_data` simply returns _all_ channels by default, and by introducing one additional argument there would be a simple way to get _all good_ channels, which is a common use case (at least for me).

But anyway, I'm fine with one extra line of code if you think it's not worth it.

agramfort, 2017-06-06:
I think it's more explicit like this. At least you know what channel
corresponds what row in data

closed 2017-06-07 by cbrnr