mne-tools/mne-python #4308: How is loose handled with volume or mixed source spaces?

joewalter, 2017-06-08:
I am wondering how the loose parameter are used with volume or mixed source spaces. It is possible to make an inverse operator with a volume source space with loose not being None. The log says "Converting to surface-based source orientations" even with volume source space. But is that really what one wants to have? I think that while using a mixed source space it could be useful to apply loose to surface based source spaces as a cortical constraint, but not to volume source spaces. But I don't see in the code that MNE discriminates between types of source spaces while applying the loose parameter. Any hints?

agramfort, 2017-06-08:
you cannot use ori and loost with a volume source space

joewalter, 2017-06-08:
But is there a test or is it set automatically anywhere? Does that imply that loose is not possible in mixed source spaces?

agramfort, 2017-06-08:
no it should crash. If it's not the case it needs a fix.

joewalter, 2017-06-08:
It is possible to run plot_mixed_source_space_inverse.py with loose=0.2 and depth=0.8 without crash.

agramfort, 2017-06-08:
is it simply ignored then?

joewalter, 2017-06-08:
The log says "Creating the depth weighting matrix..." and "Applying loose dipole orientations. Loose value of 0.2.", so it doesn't seem to be ignored.

Eric89GXL, 2017-06-08:
Dup of #2163. Keeping this open and re-titling as a bug

agramfort, 2017-06-08:
this needs a fix :(

Eric89GXL, 2017-06-08:
Sure, but whatever fix is implemented needs to be considered in the context of #2163

joewalter, 2017-06-08:
@haribharadwaj do you have any suggestions?

Eric89GXL, 2017-06-08:
I think the API proposal was `loose=dict(surf=..., discrete=..., volume=...)`. I'm okay with making `loose=float` an alias for `loose=dict(surf=float, discrete=float, volume=None)`. This is a backward-incompatible change but probably qualifies as a bugfix.

joewalter, 2017-06-09:
I'll give it a try

larsoner, 2019-06-11:
I think we have taken care of this in a recent PR, closing

closed 2019-06-11 by larsoner