mne-tools/mne-python #4309: Bug: SVD did not converge in maxwell_filter

jjnurminen, 2017-06-08:
```
Opening raw data file /home/jussi/projects/mne-python/tsss_crash_raw.fif...
    Read a total of 8 projection items:
        generated with autossp-1.0.1 (1 x 306)  idle
        generated with autossp-1.0.1 (1 x 306)  idle
        generated with autossp-1.0.1 (1 x 306)  idle
        generated with autossp-1.0.1 (1 x 306)  idle
        generated with autossp-1.0.1 (1 x 306)  idle
        generated with autossp-1.0.1 (1 x 306)  idle
        generated with autossp-1.0.1 (1 x 306)  idle
        generated with autossp-1.0.1 (1 x 306)  idle
Current compensation grade : 0
    Range : 169000 ... 469000 =     56.333 ...   156.333 secs
Ready.
Maxwell filtering raw data
    Loading raw data from disk
tsss_crash.py:14: RuntimeWarning: 102 T1/T2 magnetometer channel types found. If using SSS, it is advised to replace coil types using "fix_mag_coil_types".
  cross_talk=sss_ctc)
102 T1/T2 magnetometer channel types found. If using SSS, it is advised to replace coil types using "fix_mag_coil_types".
    No bad MEG channels
    Processing 204 gradiometers and 102 magnetometers
    Using fine calibration sss_cal_biomag_triux.dat
        Adjusting non-orthogonal EX and EY
        Adjusted coil positions by (μ ± σ): 0.3° ± 0.3° (max: 2.3°)
    Automatic origin fit: head of radius 97.5 mm
    Using origin -0.4, 10.7, 49.7 mm in the head frame
    Processing data using tSSS with st_duration=10.0
    Computing regularization
        Resulting information: 392.0 bits/sample (99.0% of peak 395.9)
        Using 72/80 inside and 15/15 outside harmonic components
    Processing 10 data chunks of (at least) 10.0 sec
        Projecting  3 intersecting tSSS components for     0.00 -    10.00 sec
        Used  1 head position for     0.00 -    10.00 sec
        Projecting  1 intersecting tSSS components for    10.00 -    20.00 sec
        Used  1 head position for    10.00 -    20.00 sec
        Projecting  1 intersecting tSSS components for    20.00 -    30.00 sec
        Used  1 head position for    20.00 -    30.00 sec
        Projecting  2 intersecting tSSS components for    30.00 -    40.00 sec
        Used  1 head position for    30.00 -    40.00 sec
Traceback (most recent call last):
  File "tsss_crash.py", line 14, in <module>
    cross_talk=sss_ctc)
  File "<string>", line 2, in maxwell_filter
  File "/home/jussi/projects/mne-python/mne/utils.py", line 609, in verbose
    return function(*args, **kwargs)
  File "/home/jussi/projects/mne-python/mne/preprocessing/maxwell.py", line 438, in maxwell_filter
    n_positions, t_str)
  File "/home/jussi/projects/mne-python/mne/preprocessing/maxwell.py", line 621, in _do_tSSS
    t_proj = _overlap_projector(orig_in_data, resid, st_correlation)
  File "/home/jussi/projects/mne-python/mne/preprocessing/maxwell.py", line 1570, in _overlap_projector
    Q_res = linalg.qr(_orth_overwrite((data_res / n).T),
  File "/home/jussi/projects/mne-python/mne/preprocessing/maxwell.py", line 1543, in _orth_overwrite
    **check_disable)[:2]
  File "/usr/local/anaconda2/lib/python2.7/site-packages/scipy/linalg/decomp_svd.py", line 119, in svd
    raise LinAlgError("SVD did not converge")
numpy.linalg.linalg.LinAlgError: SVD did not converge

```

Eric89GXL, 2017-06-08:
Can you open a PR to have it use `_safe_svd` instead of `linalg.svd`? This is a long-standing Intel MKL bug that [they know about but haven't fixed](https://software.intel.com/en-us/comment/1904961).

jjnurminen, 2017-06-08:
To reproduce:
dl & extract https://www.dropbox.com/s/lg0gi47hvoxqvzn/tsss_crash.tar.gz?dl=0
run tsss_crash.py

jjnurminen, 2017-06-08:
I fetched the latest upstream/master and it's already using `safe_svd`. So that's fixed. Thanks @Eric89GXL

closed 2017-06-08 by jjnurminen