mne-tools/mne-python #4311: read_ch_connectivity fails when channel has no neighbours defined.

fraimondo, 2017-06-09:
I computed a neighbours structure for an EGI 256 electrodes layout using fieldtrip and tried to read that mat file with `read_ch_connectivity`.

Electrode 31 in my montage has no neighbours, so the entire function failed, at two places.

1) `_recursive_flatten` assumes the `cell` parameter passed has at least one element.

This is a possible fix (just added the if condition):
``` Python
def _recursive_flatten(cell, dtype):
    """Unpack mat files in Python."""
    if len(cell) > 0:
        while not isinstance(cell[0], dtype):
            cell = [c for d in cell for c in d]
    return cell
```

2) `_ch_neighbor_connectivity` computes the set subtraction to raise an error:
`rest = set(ch_names) - set_neighbors` with the error "Some of your neighbors are not present in the list of channel names". In this case the error message and the operation do not match. The way I see it, it is an error only if the neighbours list has at least one electrode that it is not present in `ch_names`. If this is the point, then the operation should be `rest = set_neighbors - set(ch_names)`

jaeilepp, 2017-06-20:
Is there a reason this never made it to a pull request? The fixes LGTM.

fraimondo, 2017-06-22:
Not a popular issue apparently... PR coming.

PR #4337 (2017-06-24): Fix #4311
This should fix #4311
commit 07954161 (2017-06-24): Fix #4311 (#4337)

* Fix #4311

* Test for #4311

* Test for neighbours
closed 2017-06-24 by agramfort