mne-tools/mne-python #4313: loose = None is treated differently in mne and mxne

joewalter, 2017-06-09:
While loose=None with a free orientation leadfield is treated as loose=1.0, i.e., no orientation constraint, MxNE converts the forward operator to fixed orientation if loose=None, i.e., fixed orientation constraint.

Eric89GXL, 2017-06-09:
Maybe we should get rid of `loose=None` altogether and only allow floats.

joewalter, 2017-06-16:
Is it considered as a bugfix if we change the use of loose in MxNE? Wondering if we need a deprecation cycle as it's not a backward compatible fix.

Eric89GXL, 2017-06-16:
We converged on using `loose=None` as default. But maybe we should actually disallow `loose=None` for a cycle to force people to explicitly set it to 0 or 1 as necessary, since it was inconsistent (or unclear) what it was doing before. WDYT?

joewalter, 2017-06-16:
I think we should just keep it consistent, so None means do not do anything with the forward, that's why I would change it in the sparse solvers

agramfort, 2017-06-17:
I can leave with no-deprecation cycle for this corner case using mxne.
Maybe we consider it as bug fix?

just trying not to discourage @joewalter :)

joewalter, 2017-06-17:
I am totally fine with allowing only 0<=loose<=1 in the future ;-)

agramfort, 2017-06-17:
good !

larsoner, 2018-04-04:
@joewalter was this one fixed? If not, any chance to make a quick fix for 0.16?

larsoner, 2019-08-20:
I think this has been taken care of/documented at this point

closed 2019-08-20 by larsoner