mne-tools/mne-python #4318: depth weighting

annapasca, 2017-06-14:
We are attempting to add "depth weighting" to the mixed LF configuration ([#4317](https://github.com/mne-tools/mne-python/pull/4317)). But it seems python MNE does not allow for "depth weighting" in the case of fixed orientations (nor free oientation), but only in connection with the loose orientation. Do you confirm this?

We want to avoid the conversion to surface space.

Eric89GXL, 2017-06-14:
According to the docs for [make_inverse_operator](https://mne-tools.github.io/dev/generated/mne.minimum_norm.make_inverse_operator.html#mne.minimum_norm.make_inverse_operator), the combination of `loose=None, fixed=True, depth=0.8` is allowed (if the forward is in free orientation and surface-oriented). In other words, the forward should have all 3 components and be surface-oriented, but during inverse computation it will use depth weighting and change it to fixed orientation, which sounds like what you want.

Eric89GXL, 2017-06-14:
> We want to avoid the conversion to surface space.

... but you want fixed orientation for a surface? Don't you want surface orientation here?

annapasca, 2017-06-14:
We think that in the case of cortex+volume src space is not possibile to combine the surface orientation with free orientation. For this we choose to mix the LF (fixed ori for cortex + free ori for vol)

Eric89GXL, 2017-06-14:
> For this we choose to mix the LF (fixed ori for cortex + free ori for vol)

You mean fixed + surface orientation (as opposed to fixed, non-surf ori which I think would mean just the Z direction?) for cortex, w/free orientation for volume, right?

That sounds like what #4317 is trying to do. Maybe it could re-use / combine with some of this code?

Eric89GXL, 2017-06-14:
Argh I mean it sounds like what #4312 is trying to do. And maybe #4317 could share code with #4312 or vice-versa.

annapasca, 2017-06-14:
for the cortex 
```
source_nn_cortex = np.concatenate([s['nn'][s['vertno'], :] for s in fwd['src'][:2]], axis=0)
fix_rot = _block_diag(source_nn_cortex.T, 1)
LF_cortex_fixed = (LF_cortex * fix_rot).astype('float32')
```

we don't convert to surf ori

Eric89GXL, 2017-06-14:
To me that looks like code to convert to surface orientation, i.e. the first component (here the only one you use) is oriented normally to the surface, which is like what is done [here](https://github.com/mne-tools/mne-python/blob/master/mne/forward/forward.py#L587)...?

annapasca, 2017-06-14:
yes, it is! and we combine this lead field with the one for volume

Eric89GXL, 2017-06-14:
This is what #4312 is moving toward with `loose` applying only to the surface source spaces, leaving volume/discrete in free orientation.

annapasca, 2017-06-14:
We don't use loose since it has to be None for mixed src space and we added a flag to create this LF...here the code!
```
+    elif is_mixed:
+
+        vertno = [s['vertno'] for s in fwd['src']]
+        nvert = [len(vn) for vn in vertno]
+
+        # source_nn_cortex contains the normal orientation of the cortex,
+        # dim is n_source_cortex x 3
+        n_source_cortex = np.sum(nvert[:2])
+        source_nn_cortex = np.concatenate([s['nn'][s['vertno'], :]
+                                               for s in fwd['src'][:2]], axis=0)
+        print('dim of source_nn_cortex {}'.format(source_nn_cortex.shape))
+        
+        n_source_aseg = np.sum(nvert[2:])
+        source_nn_aseg = np.kron(np.ones((n_source_aseg, 1)), np.eye(3))
+        print('dim of source_nn_aseg {}'.format(source_nn_aseg.shape))
+        
+        fwd['source_nn'] = np.concatenate((source_nn_cortex, source_nn_aseg), axis=0)
+        print('dim of source_nn {}'.format(fwd['source_nn'].shape))
+        
+        n_col = n_source_cortex + n_source_aseg*3
+        
+        if n_col != fwd['source_nn'].shape[0]:
+            raise ValueError('mismatch btw ncol and fwd[''source_nn''].shape[0]')        
+        
+        fix_rot = _block_diag(source_nn_cortex.T, 1)
+        
+        LF_all = fwd['_orig_sol']
+        LF_cortex = LF_all[:, :n_source_cortex*3]        
+        LF_aseg = LF_all[:, n_source_cortex*3:]
+        
+        if (LF_all.shape[1] - LF_cortex.shape[1]) != n_source_aseg*3:
+            raise ValueError('dim mismatch!!!***####')
+        
+        LF_cortex_fixed = (LF_cortex * fix_rot).astype('float32')
+        print('LF_Cortex_fixed dim {}'.format(LF_cortex_fixed.shape))
+        
+        fwd['sol']['data'] = np.concatenate((LF_cortex_fixed, LF_aseg), axis=1)
+        print('LF all dim {}'.format(fwd['sol']['data'].shape))
+        
+        fwd['sol']['ncol'] = n_col
+        fwd['source_ori'] = 5  ## TODO choose a number to include in FIFF!!!
+        fwd['surf_ori'] = False  # TODO to be sure True or False
```

Eric89GXL, 2017-06-14:
> We don't use loose since it has to be None for mixed src space and we added a flag to create this LF...here the code!

Yes I see this is the code for #4317. I'm saying that `loose` *should* be allowed, but only apply to surface source spaces. This is why I keep pointing to #4312 (it does this / plans to do this), can you take a look and the surrounding discussion?

annapasca, 2017-06-14:
sure, we join this discussion!

annapasca, 2017-06-14:
...the last question...so do you combine a fwd with loose ori for cortex with a fwd with free ori for vol?

Eric89GXL, 2017-06-14:
Yeah that is the planned goal of #4312 I think. Would that work for you?

We haven't discussed depth weighting much there, though. Not sure if it would interact...

annapasca, 2017-06-14:
ok...now it's clear. We have only the doubt if this combination is possible in the case of loose, since in this case you rotate to a different coo system.

Eric89GXL, 2017-06-14:
> in this case you rotate to a different coo system.

Yes but you only need to rotate a subset of sources, namely the surface ones, to their own individual coordinate frames. The others (volumes) can stay in their current XYZ / FreeSurfer SurfaceRAS coordinate frame. Then we know `inv['surf_ori'] = True` refers *only to surface source spaces*, and volumes are never that way.

joewalter, 2017-06-15:
In my opinion, rotation isn't a problem for vol and discrete source spaces as we agreed to apply loose=None in #4312. So rotation has no effect.

annapasca, 2017-06-23:
Thanks @Eric89GXL @joewalter  for the explanation!

larsoner, 2019-08-20:
Closing in favor of discussion in #4312

closed 2019-08-20 by larsoner