mne-tools/mne-python #4319: BUG?: Negative WPLI

Eric89GXL, 2017-06-14:
Our WPLI can be slightly negative, but Vinck et al. Neuroimage 2011 eq. 8 suggests that it should be bounded by 0 and 1. (From the mailing list.)

Eric89GXL, 2017-06-14:
... from the code I see `np.abs` used in places that seem like they should prevent this, but who knows.

agramfort, 2017-06-14:
no idea

jstout211, 2017-06-15:
Here are output matrices from the wpli_debiased measure - they have several entries that are negative.

[dat2.npy.gz](https://github.com/mne-tools/mne-python/files/1078045/dat2.npy.gz)
[dat3.npy.gz](https://github.com/mne-tools/mne-python/files/1078046/dat3.npy.gz)

Eric89GXL, 2017-06-15:
Thanks @jstout211. I think these values are expected for debiased measures like `wpli2_debiased`. Do you have values for `wpli`? Those should be strictly [0, 1].

jstout211, 2017-06-15:
Code and 1.5seconds of data (sfreq=2000Hz, so it forced me to crop it down to 1.5s in order to upload with a 10Mb size limit)

[Small_dataset.fif.gz](https://github.com/mne-tools/mne-python/files/1078100/Small_dataset.fif.gz)

[dspm_label_connectivity.py.gz](https://github.com/mne-tools/mne-python/files/1078081/dspm_label_connectivity.py.gz)

jstout211, 2017-06-15:
No I have only been using the debiased measure.  If those are expected values, then everything should be fine.  I was not expecting a negative value.  It is likely just my error then.

Eric89GXL, 2017-06-15:
Okay I'll close for now -- see my response on the mailing list, but briefly, the debiasing is expected to produce some small negative values.

closed 2017-06-15 by Eric89GXL