mne-tools/mne-python #4323: bug when pick_ori="normal" and loose=1?

adykstra, 2017-06-16:
Should the upper bound of the following inequality (line 667 in inverse.py) be inclusive?

`is_loose = 0 < inv['orient_prior']['data'][0] < 1`

joewalter, 2017-06-16:
The problem is that with `loose = 1` we don't rotate the coordinate system. So if you used a forward operator with non-surface oriented local coordinate systems to construct your inverse then picking normal won't work. As an alternative we could decide to always rotate the coordinate system if loose is not `None`. I could add that in #4312. However this is dangerous when working with more that one source space including volume or discrete. We definitely need to check the applicability of the input parameters depending on the type of source space.

Eric89GXL, 2017-06-16:
> As an alternative we could decide to always rotate the coordinate system if loose is not None. I could add that in #4312. However this is dangerous when working with more that one source space including volume or discrete.

Yes I would say always rotate if loose is not None.

I don't see how there is any problem with vol/discrete source spaces if we say that `loose` only applies to surface source spaces...?

agramfort, 2017-06-16:
+1 to rotate if loose is not None

Eric89GXL, 2017-06-16:
@agramfort is there a good reason why we don't always rotate to surf ori for surfaces?

joewalter, 2017-06-16:
@Eric89GXL I see no problem, if we do not allow picking normal at all if there are non-surface source spaces. If picking normal. However if we allow it, then I think the line `is_loose = 0 < inv['orient_prior']['data'][0] < 1` is dangerous as it depends on the order of the src in the forward (e.g. inv['orient_prior']['data'][0] is not the loose value if the first src in the forward model is a volume).

Eric89GXL, 2017-06-16:
> e.g. inv['orient_prior']['data'][0] is not the loose value if the first src in the forward model is a volume).

Why? Can't we make `inv['orient_prior']` always be the orient prior for the surface source spaces?

agramfort, 2017-06-16:
> @agramfort is there a good reason why we don't always rotate to surf ori for surfaces?

Not always needed but i don't a real usecase

joewalter, 2017-06-16:
Isn't inv['orient_prior']['data'] just the orient_prior computed in compute_orient_prior?

Eric89GXL, 2017-06-16:
I don't know without looking at the code, I believe you that it is.

In any case, can we do something better than just using `inv['orient_prior']['data'][0]` to alleviate your concern / the potential danger? I don't quite see the blocker here, maybe some code snippets / use cases would help.

joewalter, 2017-06-16:
Using the minimum should be enough

larsoner, 2021-02-19:
Looking at the code more recently, I think we're stuck with our existing convention of `< 1` rather than `<=1` for backward compat purposes so I'll close this

closed 2021-02-19 by larsoner