mne-tools/mne-python #4327: GDF Files NaNs read as lowest negative number + other  data 1e6 smaller

robintibor, 2017-06-19:
When reading GDF files, in some cases NaNs are converted to the lowest negative number, is this intended behavior?

```python
>>> raw_edf = mne.io.read_raw_edf('/home/schirrmr/data/bci-competition-iv/2b/B0101T.gdf')
>>> data = raw_edf.get_data().T
>>> data[52250:52352,:3] # there should be nans within that range
array([[ -2.13321126e-06,  -2.71305409e-06,  -6.13412680e-07],
       [ -1.00001526e-04,  -1.00001526e-04,  -1.00001526e-04],
       [ -1.00001526e-04,  -1.00001526e-04,  -1.00001526e-04],
       [ -1.00001526e-04,  -1.00001526e-04,  -1.00001526e-04],
       [ -1.00001526e-04,  -1.00001526e-04,  -1.00001526e-04],
       [...]
       [ -1.00001526e-04,  -1.00001526e-04,  -1.00001526e-04],
       [ -1.00001526e-04,  -1.00001526e-04,  -1.00001526e-04],
       [ -1.00001526e-04,  -1.00001526e-04,  -1.00001526e-04],
       [ -1.00001526e-04,  -1.00001526e-04,  -1.00001526e-04],
       [  3.66216526e-08,   2.79240101e-06,  -3.98260472e-06]])
```

Dataset is from this: http://www.bbci.de/competition/iv/#dataset2b
The description given there about NaNs is:

> Note that the runs are separated by 100 missing values, which are encoded as not-a-numbers (NaN) by default.  Alternatively, this behavior can be turned off and the missing values will be encoded as the negative maximum values as stored in the file with [...]

But I didn't see this behavior (negative minimum value) documented in mne?
Also, at least on my PC, these minimum values are not identical for different channels, which could cause confusion?

```python
>>> np.min(data[:,:3])
-0.00010000152590218967
>>> np.min(data)
-0.0010000152590218965
```

Also, non-NaN values are factor 1e6 smaller than the values I get from biosigtoolbox in matlab.

jaeilepp, 2017-06-20:
I looked at the data and indeed, the values are negative maximum values (int16, -32768). So nothing is actually converted, the data just is encoded to be small, which turns into some other small values when calibrated accordingly. You can work around this by setting the small values to NaNs yourself using some threshold. The small differences are probably caused by floating point imprecisions.

> Also, non-NaN values are factor 1e6 smaller than the values I get from biosigtoolbox in matlab.

Biosigtoolbox is is probably presenting the data as microvolts. In MNE the EEG data are in volts.

jaeilepp, 2017-06-22:
I think this is just a feature of Biosigtoolbox in how it treats minimal values. Closing.

closed 2017-06-22 by jaeilepp

robintibor, 2017-06-22:
Thanks for the explanations :)