mne-tools/mne-python #4328: Why can't I plot two topomaps side by side in a Jupyter notebook?

cbrnr, 2017-06-20:
I would like to plot two topomaps side by side in a Jupyter notebook. However, the following code (which works as expected in a Python console) only plots one topomap and an empty figure when run in a notebook:

```Python
import numpy as np
import matplotlib.pyplot as plt
import mne

chans = ['Fp1', 'AF7', 'AF3', 'F1', 'F3', 'F5', 'F7', 'FT7',
         'FC5', 'FC3', 'FC1', 'C1', 'C3', 'C5', 'T7', 'TP7',
         'CP5', 'CP3', 'CP1', 'P1', 'P3', 'P5', 'P7', 'P9',
         'PO7', 'PO3', 'O1', 'Iz', 'Oz', 'POz', 'Pz', 'CPz',
         'Fpz', 'Fp2', 'AF8', 'AF4', 'AFz', 'Fz', 'F2', 'F4',
         'F6', 'F8', 'FT8', 'FC6', 'FC4', 'FC2', 'FCz', 'Cz',
         'C2', 'C4', 'C6', 'T8', 'TP8', 'CP6', 'CP4', 'CP2',
         'P2', 'P4', 'P6', 'P8', 'P10', 'PO8', 'PO4', 'O2']

info = mne.create_info(chans, sfreq=512, ch_types="eeg", montage=mne.channels.read_montage("biosemi64"))

fig, (left, right) = plt.subplots(ncols=2, figsize=(10, 5))
mne.viz.plot_topomap(np.random.randint(0, 100, 64), info, vmin=0, axes=left)
left.set_title("x1")
mne.viz.plot_topomap(np.random.randint(0, 100, 64), info, vmin=0, axes=right)
right.set_title("x2")
fig.suptitle("Overall title")
```
[image]

Any ideas why this is happening and how I could fix it?

jaeilepp, 2017-06-20:
I'm not very fluent with notebooks, but I think the problem is that the figure gets drawn prematurely. So hopefully it's fixed by changing last lines to:
```
fig, (left, right) = plt.subplots(ncols=2, figsize=(10, 5))
mne.viz.plot_topomap(np.random.randint(0, 100, 64), info, vmin=0, axes=left, show=False)
left.set_title("x1")
mne.viz.plot_topomap(np.random.randint(0, 100, 64), info, vmin=0, axes=right, show=True)
right.set_title("x2")
fig.suptitle("Overall title")
```

cbrnr, 2017-06-20:
Nice, thank you, this worked! 🎉

closed 2017-06-20 by cbrnr

jaeilepp, 2017-06-20:
Maybe switching to interactive mode also fixes this without having to use the show param. It looks like the titles don't get drawn either, so one option is to plot everything with ``show=False`` and just call ``plt.show()`` at the end.

cbrnr, 2017-06-20:
Unfortunately, `%matplotlib inline` doesn't help either (it seems to be the default anyway). Yes, you're right, setting `show=False` everywhere is the best option.

mmagnuski, 2017-06-20:
(interactive mode should be `%matplotlib` - where the plot is in a separate figure which you can interact with while still being able to interact with the notebook)

cbrnr, 2017-06-20:
@mmagnuski you're right, but that's generally not what I'd want when using a notebook (but of course there is `%matplotlib notebook` IIRC) - so delaying all drawing operations until the end of the cell is a perfect solution!