mne-tools/mne-python #4340: Support for events in EDF files

cbrnr, 2017-06-23:
I ran into two issues related to loading EDF files.

1. The default for `stim_channel` is `-1`, but what if there is no stim channel in the file? I was expecting that this should be detected, because other toolboxes (such as Biosig) automatically detect if there is a stim channel in the file. An example EDF file without a stim channel can be found [here](http://www.teuniz.net/edf_bdf_testfiles/test_generator.zip).
2. It seems that events are not loaded/parsed correctly in some EDF files. For example, [this EDF+ file](http://www.teuniz.net/edf_bdf_testfiles/test_generator_2_edfplus.zip) contains 2 events (at the very first and last samples). `mne.find_events` doesn't find any events - the stim channel signal contains a value of `2.0` for the first sample, but all other samples are `0.0`. So it can't be just using the wrong arguments for `mne.find_events`.

cbrnr, 2017-06-23:
It seems like [pyedflib](http://pyedflib.readthedocs.io/en/latest/) correctly reads these files.

jaeilepp, 2017-06-23:
>The default for stim_channel is -1, but what if there is no stim channel in the file? I was expecting that this should be detected, because other toolboxes (such as Biosig) automatically detect if there is a stim channel in the file. An example EDF file without a stim channel can be found here.

I don't think there's a way of automatically detect the stim channel in edf, since this info does not exist in the meta data. The only way I see is to guess using the ch_name or unit, but that seems dangerous. So changing ``stim_channel`` default to None would fix this, but in my experience -1 works quite often and changing it would require deprecation. Maybe adding a warning in the docs would suffice.

> It seems that events are not loaded/parsed correctly in some EDF files. For example, this EDF+ file contains 2 events (at the very first and last samples). mne.find_events doesn't find any events - [...]

Since the ``tal_channel`` gets parsed to construct a stim channel, and ``find_events`` detects rising edges, they aren't detected. The start is not detected because the stim channel starts at non zero, and the end is not detected because the rising edge would be at sample n+1. We could change ``find_events`` to detect non zero starts, but that could have side effects. Another option would be to have ``get_edf_events`` method that would return the parsed tal_channel.

cbrnr, 2017-06-23:
> I don't think there's a way of automatically detect the stim channel in edf, since this info does not exist in the meta data. The only way I see is to guess using the ch_name or unit, but that seems dangerous.

But how are other libs such as Biosig or pyedflib handling this? They seem to be able to detect events automatically. This would be very convenient, because right now the user has to jump through some hoops to find out if there are events in the file.

>Since the tal_channel gets parsed to construct a stim channel, and find_events detects rising edges, they aren't detected.

OK, I have no idea what the `tal_channel` is, but it seems event information is present in this special(?) channel. Is this channel always present? Can it be addressed independent of its name? I'm sorry, but as I said I don't know much about the EDF format.

>The start is not detected because the stim channel starts at non zero, and the end is not detected because the rising edge would be at sample n+1. We could change find_events to detect non zero starts, but that could have side effects.

Plus it would only detect the first event. How can the second event be detected at all?

>Another option would be to have get_edf_events method that would return the parsed tal_channel.

If this is a more generic solution and if this solves the issue of automatically detecting if there are events at all, then this would be perfect!

jaeilepp, 2017-06-23:
> But how are other libs such as Biosig or pyedflib handling this? They seem to be able to detect events automatically. This would be very convenient, because right now the user has to jump through some hoops to find out if there are events in the file.

AFAIK the old EDF format has no events encoding as such. I suppose the other libs just read the data as it is, not even trying to find the events (bullet 1. in the original post).

> OK, I have no idea what the tal_channel is, but it seems event information is present in this special(?) channel. Is this channel always present? Can it be addressed independent of its name? I'm sorry, but as I said I don't know much about the EDF format.

In EDF+ the tal channel contains the [annotations](http://www.edfplus.info/specs/edfplus.html#edfplusannotations) (events, bullet point 2.).

> Plus it would only detect the first event. How can the second event be detected at all?

Yes. I don't think there is a clean solution for this without breaking something else.

> If this is a more generic solution and if this solves the issue of automatically detecting if there are events at all, then this would be perfect!

I'll take a stab at this.

cbrnr, 2017-06-23:
> AFAIK the old EDF format has no events encoding as such. I suppose the other libs just read the data as it is, not even trying to find the events (bullet 1. in the original post).

OK, so is there a flag somewhere that tells me that what I've loaded is an old EDF file? That way I could call `mne.find_events` with `stim_channel=None`.

> I'll take a stab at this.

Cool - is using `pyedflib` an option to simplify things and get maximum EDF compatibility?

jaeilepp, 2017-06-23:
I guess the only way is to check whether 'EDF Annotations' exists in the channels. I'm thinking that maybe we should switch to ``stim_channel='auto'`` by default like @Eric89GXL suggested earlier. The only problem is that if there's a channel with that name, we have a conflict. I guess we could just raise an error in that case.

cbrnr, 2017-06-23:
Would this solve the issue of automatically detecting events in any EDF file (old or new)?

If there really is a channel called `'auto'` and you want to designate it as the stim channel, how would you do it?

jaeilepp, 2017-06-23:
> Would this solve the issue of automatically detecting events in any EDF file (old or new)?

If 'auto', it would assign a stim channel if 'EDF Annotations' or GDF events exist. Otherwise it would not assign a stim channel.

> If there really is a channel called 'auto' and you want to designate it as the stim channel, how would you do it?

You could pass the index of the channel.

cbrnr, 2017-06-23:
> If 'auto', it would assign a stim channel if 'EDF Annotations' or GDF events exist. Otherwise it would not assign a stim channel.

Nice, so this solution would be independent of specific names for stim channels. The behavior would correspond to what Biosig or pyedflib are doing. I like that a lot!

> You could pass the index of the channel.

Fair enough, if this is mentioned in the docs then I don't see any problem with it!

cbrnr, 2017-06-23:
Come to think of it, GDF stores events in an event table (annotations). Does that mean that MNE always converts events first into a stim channel and then parses that stim channel to get events?

jaeilepp, 2017-06-23:
Yes.

closed 2017-06-23 by agramfort

cbrnr, 2017-06-23:
Thanks @jaeilepp, this is a great improvement! Just to make sure, this fixes the first issue I mentioned above, but what about parsing the stim channel correctly in the case of [this EDF+ file](http://www.teuniz.net/edf_bdf_testfiles/test_generator_2_edfplus.zip)? Should I open a new issue since this is separate from the current fix?

jaeilepp, 2017-06-26:
Yeah, in that case the stim channel should actually be appended, not assigned. I'll fix it.

jaeilepp, 2017-06-26:
Or actually, I'm not sure. I don't see why someone would want to have the 'EDF  Annotations' channel present, since the data does not make any sense without parsing.

Regarding the start and stop events, I thought the PR took care of it. I'd prefer not to mess with ``find_events`` function, since that'll break things. Maybe just mention something in the docstring.

cbrnr, 2017-06-26:
@jaeilepp I'm not sure what you meant with your last comments regarding the stim channel (appended vs. assigned). Having the events is important, the annotations channel is just an internal representation of events that the user doesn't need to see.

Re start and stop events, no, `mne.find_events` still doesn't find any of these two events. Actually, the stim channel still looks the same, with all zeros except for the first sample. With this information, I think the last event cannot even be reconstructed. Since the events are clearly present in the file (as e.g. loaded by other toolboxes), `mne.find_events` is already broken in this case.

jaeilepp, 2017-06-26:
> Re start and stop events, no, mne.find_events still doesn't find any of these two events. Actually, the stim channel still looks the same, with all zeros except for the first sample.

Yes, the events are outside the data range. This is what the new function is for.

cbrnr, 2017-06-26:
Nice! Sorry, I somehow didn't see the new function! Thanks @jaeilepp!

cbrnr, 2017-11-14:
I'm sorry I have to revive this issue, but I'm still running into trouble regarding EDF events. I thought that the new argument `stim_channel='auto'` correctly handled EDF files with events stored as annotations. However, [this file](https://www.physionet.org/pn4/eegmmidb/S001/S001R04.edf) still gets an extra stim channel when loaded with `read_raw_edf` even though it has event annotations. In this specific case, this means that `mne.find_events` reports fewer events than there are actually present - the first event is missing (because it occurs at time 0).  The new `find_edf_events` method correctly returns the events (but in a different format not compatible with the array returned by `mne.find_events`).

In short, loading EDF files and parsing events is still a mess. It would be nice if there was one consistent workflow to get events. Maybe an option for `mne.find_events` to consider a non-zero first sample (so without a rising/falling edge) as an event would be a good start.

@larsoner WDYT?

cbrnr, 2017-11-14:
Update: On second thought, I think it is OK that such files get an added stim channel, but the problem is still in parsing this stim channel.

larsoner, 2017-11-14:
> The new find_edf_events method correctly returns the events (but in a different format not compatible with the array returned by mne.find_events).

IIRC the events that are stored in EDF contain extra information (maybe duration?) so it necessitated using a different format. Maybe a doc update to say that this is intentional and giving a little 3-line NumPy recipe for how to convert to standard `events` array would be useful.

> Maybe an option for mne.find_events to consider a non-zero first sample (so without a rising/falling edge) as an event would be a good start.

Sure, a `initial_event=False` (backward-compat default) or so seems reasonable, I think this has come up before.

cbrnr, 2017-11-14:
> IIRC the events that are stored in EDF contain extra information (maybe duration?) so it necessitated using a different format. Maybe a doc update to say that this is intentional and giving a little 3-line NumPy recipe for how to convert to standard events array would be useful.

You're right, it seems like EDF events have durations, so it is good to be able to get this information somehow. I'll try to come up with a short recipe, and we should definitely update the docstring to describe the output of `find_edf_events`.

> Sure, a `initial_event=False` (backward-compat default) or so seems reasonable, I think this has come up before.

OK, I'll create a PR soon.

But for the user who wants to load an EDF file and has no idea how events are stored - how do we make sure that MNE makes doing the right thing as simple as possible? It would be nice to just say "use `mne.find_events` to find events, no matter what the file format is". But I guess we could issue a warning (or info?) if `read_raw_edf` detects event annotations?

larsoner, 2017-11-14:
> It would be nice to just say "use mne.find_events to find events, no matter what the file format is". But I guess we could issue a warning (or info?) if read_raw_edf detects event annotations?

Ideally we can say just use `mne.find_events`.Users who do know about the events they expect from an EDF specifically might be disappointed because there are no durations, which is why `find_edf_events` exists.

We could do `initial_event=True` by default via deprecation, but how common are such events in EDF files? My initial idea is to have `False` by default, and document it properly in the docstring. Then hopefully people will realize that if they tend to have events right at the recording onset, they should use `True`.

cbrnr, 2017-11-14:
> Ideally we can say just use `mne.find_events`. Users who do know about the events they expect from an EDF specifically might be disappointed because there are no durations, which is why `find_edf_events` exists.

Can we log an info if `read_raw_edf` detects event annotations?

> We could do `initial_event=True` by default via deprecation, but how common are such events in EDF files?

Not that uncommon I'd say. How much would it break if we really set that to `True`? We could implement a check that even if `initial_event=True` but the first sample of the stim channel is actually 0, no event would be detected.

We could also issue a warning/info if `find_events` detects the first sample to be not equal to 0.

> My initial idea is to have False by default, and document it properly in the docstring. Then hopefully people will realize that if they tend to have events right at the recording onset, they should use True.

Maybe, if we add these warnings/info messages. Depending how much existing stuff we'd break, I'd tend to set it to `True` by default though.

larsoner, 2017-11-14:
> Can we log an info if read_raw_edf detects event annotations?

Yes, we should add it if we don't already

> How much would it break if we really set that to True? 

Users could easily know the structure of their data (and the behavior of `find_events`) and reasonably do:
```
events = find_events(...)
# fix event 1
events[0, 2] = 20
...
```
And now we've broken their code. So we'd need a deprecation cycle to avoid it. The question is, is it worth it? `find_events` is a very commonly used function (`git grep "find_events" | wc -l` in the git root itself gives 213 results) so I'm inclined to avoid a cycle if we can.

> We could implement a check that even if initial_event=True but the first sample of the stim channel is actually 0, no event would be detected... We could also issue a warning/info if find_events detects the first sample to be not equal to 0.

My idea was that it would just *enable* finding events at `t=0`, not *force* an event at `t=0` (I'm imagining an EDF user with events sometimes at zero and sometimes not). So I don't think we'd need any warning for this mode if we document it properly.

cbrnr, 2017-11-14:
- [ ] issue info if `read_raw_edf` detects event annotations

In your example, we would only break things if there was indeed an event at sample 0.

I know that `find_events` is often used, but how often in combination with EDF? Not 213 times I imagine.

So the question for me is still how much would that really break, but maybe this is difficult to answer so I'm also fine with setting the new arg to `False` (at least for now).

> My idea was that it would just enable finding events at t=0, not force an event at t=0.

Agreed.

larsoner, 2017-11-14:
> In your example, we would only break things if there was indeed an event at sample 0.

Yes but we don't know how common this is. I suspect it's not totally uncommon. For example, a subject holding a button when you press "record" during Neuromag acquisition would be enough to do it. The risk of silent breakage is too high.

> I know that find_events is often used, but how often in combination with EDF? Not 213 times I imagine.

Yes but we wouldn't just be breaking EDF users' code. Data from all systems get channeled through `find_events`, so a deprecation cycle there means emitting warnings to almost every user of MNE.

cbrnr, 2017-11-14:
Of course, I'd never *silently* break anything, but flooding users with warnings is also not ideal. OK, so do you agree with my checklist above?

larsoner, 2017-11-14:
Yeah it looks good to me

cbrnr, 2017-11-15:
Regarding the problem with events at t=last_samp, I'm not sure if the stim channel gets constructed correctly. Shouldn't the last sample in the converted stim channel have the event value? Why did @jaeilepp say that the rising edge would be at sample n+1? I think there might be something wrong with this.

cbrnr, 2017-11-15:
Or in this example, are the event annotations incorrect? They are:

    [[0.0, 0, 'RECORD START'], [600.0, 0, 'REC STOP']]

But the total length is 600s (120000 samples), which means that if time starts from 0 the last sample is actually 600-1/200=599.995 (because the sampling frequency is 1/200=0.005). This of course would mean that everything MNE does is correct.

larsoner, 2017-11-15:
Maybe he meant that the *falling* edge would be at sample `n+1`. But yes I agree that output looks fine (thinking of STOP as the first time point to exclude at least)

cbrnr, 2017-11-15:
Well this is certainly a valid explanation for a Pythonista, but I doubt that this is true in general. What if the event at t=600s was called something else? Do you agree that t=600s is an invalid time for an event in this data set?

larsoner, 2017-11-15:
> Do you agree that t=600s is an invalid time for an event in this data set?

As on onset time, yes. As an offset ... depends on how we want to define onsets and offsets. I guess we usually use inclusive numbering, which would argue for 599.995.

Do we choose to set it as 600s (e.g., by doing `samp_start + duration + 1`) or is it encoded that way in the file? If the latter, then I don't think we should change it.

cbrnr, 2017-11-15:
The events from EDF annotations are always given in terms of onsets (so 600s is invalid). I'm *pretty sure* it's encoded like this in the file, and I wouldn't change anything in this case either because it's not MNE's fault if an invalid event cannot be recovered. But I'll try to see how other tools (pyedflib, biosig) handle this.

cbrnr, 2017-11-15:
Yeah, there's something wrong with this last event; the BioSig toolbox assigns sample 12001 to this event, but at the same time the signal is indexed from 1 to 12000 (remember we're in MATLAB 1-based indexing). So I guess EDF events are not validated and can contain any value.

larsoner, 2017-11-15:
It's also possible that the STOP event is special

cbrnr, 2017-11-15:
Sure, but then BioSig also handles it incorrectly, and `pyedflib` also says that the event occurs at t=600s in a file of duration 600s (or 12000 samples), which means that t=600s does not exist if we start at t=0s. I think it's more likely that this event is simply wrong and should really be 599.995s.