mne-tools/mne-python #4343: Lack of mne-python/mne/channels/data/neighbors/ *.mat file for EEG 10-20 system based on 21 electrodes.

piotrdzwiniel, 2017-06-23:
When using `mne.channels.read_ch_connectivity` function I realize the lack of appropriate connectivity structure for EEG 10-20 system consisted of 21 electrodes with appropriate naming conventions as described for example at https://en.wikipedia.org/wiki/10%E2%80%9320_system_(EEG). Thus, I've prepared such file and it would be great if you could include it in 'mne-python/mne/channels/data/neighbors/' folder. PS It would be great if someone also could check the validity of assumed between-sensor adjacency in the below structure.

[eeg1020_21channels.zip](https://github.com/mne-tools/mne-python/files/1098219/eeg1020_21channels.zip)

Eric89GXL, 2017-06-23:
I think eventually we want to construct these on the fly using channel positions and Delaunay triangulations. Are you up for trying that?

Eric89GXL, 2017-06-23:
cc @jona-sassenhagen 

Related:

https://docs.scipy.org/doc/scipy-0.19.0/reference/generated/scipy.spatial.Delaunay.html https://docs.scipy.org/doc/scipy-0.19.0/reference/generated/scipy.spatial.ConvexHull.html

jona-sassenhagen, 2017-06-23:
I'll try to finish up my example ... It's mostly done, but I'm so slow these days.

agramfort, 2017-06-24:
@piotrdzwiniel is this file yours? did you take it from fieldtrip? is it a standard name for the file? if so +1 to add it.

mmagnuski, 2017-06-24:
Hi Piotr!
Just as @Eric89GXL says I think we could also extend mne's functionality related to adjacency matrices - @jona-sassenhagen recently posted a short notebook example of how to create such matrices with Delunay triangulation on channel positions projected to 2d. @jona-sassenhagen - by finishing the example do you also mean adding a function to construct adjacency?
I have had a function for ploting  adjacency in 3d for some time but recenty added interactivity similar to plot_neighbours in fieldtrip (in fieldtrip that's `enableedit`) - so that adjacency can be corrected after automatic construction. We could work on a PR together if you want.

piotrdzwiniel, 2017-06-27:
@Eric89GXL Yeah, no problem - I can create between-sensor adjacency matrix based on electrodes' positions and Delaunay Triangulations, in order to make the matrix more accurate and reliable. 

@agramfort  Yep, this file is created by me. It was lacking in FieldTrip. I don't know if it's a standard name for a file, but the naming of matlab workspace variables closed in it are with respect to `mne.channels.read_ch_connectivity` function, so you can use this *.mat file for the creation of connectivity matrix needed for further analyses, eg. for `mne.stats.spatio_temporal_cluster_test`.

@mmagnuski Can you provide a link to the mentioned notebook of jona's (@jona-sassenhagen)? What you mean by "PR"? Nevertheless, the answer is - sure.

mmagnuski, 2017-06-27:
@piotrdzwiniel 
PR stands for Pull Request: proposed changes to the codbase, to be reviewed and accepted by maintainers of the software. You have a tab at the top that shows all pending pull reuqests for mne.
There is currently [one adjacency-related PR open - take a look](https://github.com/mne-tools/mne-python/pull/4356). I can't find Jona's example notebook but the PR mentioned above implements what was proposed in the example.
I'll be submitting a PR for plotting (and later interactive plotting) of adjacency matrices based on code I've been using so far - you can join me if you want, either in Warsaw or remotely.

piotrdzwiniel, 2017-06-28:
@mmagnuski Ah, ok. To be honest I believe it would be better to create a separate PR for the function calculating between-sensor adjacency based on Delaunay Triangulations only. It will be much more clear and intuitive. Do you have maybe a suggestion based on your experience which mne sub-package will be best suited for this kind of functionality?

mmagnuski, 2017-06-28:
@piotrdzwiniel - I was talking about visualisation because adjacency creation in the works [here](https://github.com/mne-tools/mne-python/pull/4356).

piotrdzwiniel, 2017-06-28:
@mmagnuski Hmm, yeah... Anyway, goodluck with the visualization functionality.

Eric89GXL, 2017-06-28:
So we should add this file, and also have @mmagnuski's PR, then, right?

mmagnuski, 2017-06-28:
I'll open one soon just as a sketch  to discuss API and will wait for @jaeilepp to get merged.

larsoner, 2018-04-04:
We have adjacency creation now so I'll close this, feel free to reopen if there are bits left to do

closed 2018-04-04 by larsoner