mne-tools/mne-python #4350: Graphical frontend for MNE

cbrnr, 2017-06-26:
I noticed that many people prefer a GUI for EEG processing instead of writing scripts or using the interactive Python interpreter. Therefore, I created a little graphical frontend for MNE based on PyQt5 to make MNE's functionality accessible via a GUI. Of course this is modeled a bit after EEGLAB, which has a nice GUI especially suitable for beginners. I will be expanding this GUI with new functions and I'd be happy to hear your thoughts/comments/suggestions.

https://github.com/cbrnr/mnelab

Eric89GXL, 2017-06-26:
That's neat! Do you plan to expand the functionality in various ways?

jasmainak, 2017-06-26:
Perhaps also worth taking a look at what Jaakko had developed: https://github.com/jaeilepp/eggie

cbrnr, 2017-06-26:
Yes, I want to cover a basic EEG processing workflow similar to what can be done in EEGLAB. This should be ready soon since I want my students to switch to MNE. Thanks @jasmainak for pointing out this existing project. I see that it hasn't been updated for quite some time, but maybe some features can be ported over to MNELAB.

mmagnuski, 2017-06-26:
Would be really helpful if it could generate history scripts similar to eeglab's `eegh` - that eases up transition from GUI to code. I also see that you are using `PyQt5` which unfortunatelly doesn't work with mayavi (at least for now). But that shouldn't be problematic if you are planning to cover the basics for now.

BTW I tried teaching mne two times already but teaching `coding + mne + signal analysis` turned out to generate some confusion. I have to retreat to teaching brainstorm now, but will be watching and rooting for mnelab. :)

cbrnr, 2017-06-26:
Yes, generating code from GUI click history is definitely high on the list :smile:. I'll stick with PyQt5 for now because everything else is deprecated. I might switch it out for [QtPy](https://github.com/spyder-ide/qtpy) at some point if people really need the old stuff and if there is not much overhead. Isn't mayavi one of the few packages that still don't work with Python 3 (besides PsychoPy, which is slowly getting there 😛 )? Anyway, this is a feature for much later especially since I don't have a use case for it now.

> BTW I tried teaching mne two times already but teaching coding + mne + signal analysis turned out to generate some confusion.

I hear you, this is exactly what I learned when working with students with no background in CS. A GUI helps to lower the initial entry bar, and in many cases it is not necessary to write scripts (see EEGLAB or PsychoPy). But of course for reproducible science we always have the full power of Python scripting available to us.

> I have to retreat to teaching brainstorm now, but will be watching and rooting for mnelab. :)

Ouch! 🤕 😉

mmagnuski, 2017-06-26:
> Isn't mayavi one of the few packages that still don't work with Python 3 (besides PsychoPy, which is slowly getting there 😛 )?

Mayavi has been working with python 3 for some time but py3-compatible version was not in the official conda build so me and many other people used mayavi from menpo conda channel. But unfortunatelly mayavi doesn't work with PyQt5 yet (https://github.com/enthought/mayavi/issues/448).

> Ouch! 🤕 😉

Actually brainstorm is pretty cool, much better gui-wise than eeglab at least. But, of course, my heart is where mne-python is. :)

cbrnr, 2017-06-27:
> Actually brainstorm is pretty cool, much better gui-wise than eeglab at least. But, of course, my heart is where mne-python is. :)

I know, for a MATLAB-based GUI it is really fantastic. But it remains a MATLAB-based GUI nonetheless...

cbrnr, 2017-06-29:
Please feel free to create issues at https://github.com/cbrnr/mnelab if you have suggestions. The basic workflow is almost done, I just need to think about how to represent data that is not stored in a file (e.g. data manipulated after loading such as with filtering). The command history is pretty simplistic at the moment, but maybe this is sufficient.

cbrnr, 2017-07-12:
Data set handling is now more or less done, so time to add more features!

agramfort, 2017-07-12:
can you write here quickly your plan? just to better understand the vision.

thx for doing this

cbrnr, 2017-07-12:
My plan is to create a GUI for MNE to make it accessible for people who can't or don't want to program (and even if you can program, sometimes it is just convenient to click around and produce various standard plots/analyses). Most of my students here do not have a programming background, but I want them to perform EEG analysis on their own. We've used EEGLAB, but I hate that it is based on MATLAB (for the obvious reasons I won't discuss here), so I decided that we need a Python-based solution. EEGLAB has a large user base, and I'm pretty sure that many users would to switch to Python if there was something similar for EEG analysis just because MATLAB is so inconvenient.

For now, I mostly need very basic stuff like visualization of raw data, re-referencing, identification of bad channels, creating manual annotations, plotting the PSD, computing ICA, and removing artifact ICs. I've tried to write scripts for my students, but it doesn't work - it is very difficult for them to change even the smallest detail. They are comfortable with the EEGLAB GUI though, so I'm trying to keep MNELAB very similar to EEGLAB intentionally.

I think a GUI would attract many more people to use MNE, so once this gets going I'm hoping for contributions from others to implement stuff I don't need in the GUI.

Eric89GXL, 2017-07-12:
> For now, I mostly need very basic stuff like visualization of raw data, re-referencing, identification of bad channels, creating manual annotations, plotting the PSD, computing ICA, and removing [...]

Do you have some plan for keeping a record of what was done? The advantage to scripting is that there is some record of the serial processing steps. Will your GUI make this record and execute it interally or something, or is such a record lost?

> I think a GUI would attract many more people to use MNE, so once this gets going I'm hoping for contributions from others to implement stuff I don't need in the GUI.

I know many people who would be very happy to be able to get some of these basic things done via GUI.

cbrnr, 2017-07-12:
> Do you have some plan for keeping a record of what was done? The advantage to scripting is that there is some record of the serial processing steps. Will your GUI make this record and execute it interally or something, or is such a record lost?

Yes, all steps carried out in the GUI will be recorded in a history (list of MNE commands). It is already implemented, the challenge is to map what I'm doing internally to generic MNE commands that can be used in a standard script. But this should be possible.

> I know many people who would be very happy to be able to get some of these basic things done via GUI.

Good to hear that this is not just my feeling - I was surprised that such a GUI does not exist, so creating it was the logical step.

agramfort, 2017-07-12:
thanks for taking the time to explain the vision.

if you can make a short screencast to illustrate what you can do with
mnelab please share.

we could also host the mnelab project on the mne-tools org.

cbrnr, 2017-07-13:
I'd like to add a couple of features before I create a screencast - right now, you can't really do all that much. The same applies to migrating to mne-tools - I'd be very happy to do that, but not quite yet - give me a bit more time until MNELAB can actually be used in a simple workflow. For that, I need at least working implementations of re-referencing, support for annotations (including import and export), and artifact removal via ICA.

chrismullins, 2017-07-21:
+1 for this!  I'll try to hack on this a little bit, and add some issues over on that project for specific EEGLAB-like features to work on so we don't duplicate efforts.

cbrnr, 2017-07-24:
Happy to coordinate! I'll try to summarize the basic data structures in the program in a short markdown document to kickstart the discussion.

teonbrooks, 2017-08-17:
👍

jjnurminen, 2017-09-29:
Looks cool - I have also seen many people who absolutely will not get into Python, but would benefit from having some of the mne-python functionality available.
As a suggestion, having myself developed some modest PyQt applications, I would really recommend learning Qt Designer.

larsoner, 2018-04-04:
Closing since there is no further "todo" here, but feel free to continue to discuss

closed 2018-04-04 by larsoner

cbrnr, 2018-04-05:
Yes, all discussions related to the GUI should take place in the MNELAB repo. Currently, I'm still stuck because of https://github.com/cbrnr/mnelab/pull/15, where I need to run the ICA calculation in a separate thread to be able to show a Cancel button in the GUI... Any help is very much appreciated.

MatteoCarlone, 2021-03-25:
Can i pass from code to the gui?

drammock, 2021-03-25:
@MatteoCarlone questions about MNELAB should be raised in the MNELAB repository: https://github.com/cbrnr/mnelab/issues