mne-tools/mne-python #4352: What does set_eeg_reference() do?

nbara, 2017-06-26:
Hi all, 

I'm a bit puzzled as to what `set_eeg_reference()` actually does.

Here is a gist where I do `set_eeg_reference(ref_channels=[])` (as my data was already referenced).

https://gist.github.com/nbara/972eefa4a62fd6ef9df308b76ffa2559

Contrary to my expectations, this seems to rereference to the mean, and also appears to include the STIM channel in the mix (visible peaks every .5 seconds)??

Eric89GXL, 2017-06-26:
Based on the [docstring](https://mne-tools.github.io/dev/generated/mne.set_eeg_reference.html#mne.set_eeg_reference) it seems like doing `[]` should indeed not affect the data, but clearly it did. So I think this is a bug.

Can you take a minute to use `np.random.RandomState(0).randn(...)` as the data to turn this into a minimal example?

Are you up for digging into the code to try to fix it?

nbara, 2017-06-27:
Hi Eric,

Here's a minimal example using random data:
https://gist.github.com/nbara/8e6e2a3dbb18675b7b57fca3fbfbc111

Does that work?

I'm afraid I have too much on my plate atm so I won't be able to fix this myself...

wmvanvliet, 2017-06-27:
What the.... ok, I'm looking into it.

wmvanvliet, 2017-06-27:
Turns out, the unit test for this case was broken. This allowed me to at some point introduce a breaking change to the `set_eeg_reference()` function. My apologies.

PR #4357 (2017-06-29): [MRG] Fix referencing when ref_channels=[]
Due to faulty control-flow logic, doing `raw.set_eeg_reference(ref_channels=[])` actually computed a common-average reference. This fixes the control-flow and the botched unit test that was supposed to prevent this from happening.

closes #4352
commit 8ab4ac83 (2017-06-29): Fix referencing when ref_channels=[] (#4357)

closes #4352
closed 2017-06-29 by Eric89GXL