mne-tools/mne-python #4353: FIX: Filtering with annotations

Eric89GXL, 2017-06-26:
We currently don't account for annotations while filtering, i.e. we treat them as usable data and filter the entire time-course. Does anyone know how other packages with BAD segments deal with this problem?

One option is to filter each good segment as an independent signal (i.e., with padding and all that good stuff). This creates some questions, like what do you do if your good section is too short to reasonably filter, e.g. you have 1 sec of good data between BAD segments and you want to do a 0.5 Hz highpass. Do we mark that segment as `filter-BAD`? Or emit a warning? Or make this optional?

This type of segmented processing will have the nice side-effect of fixing nasty edge artifacts we currently get from concatenating raw files, too.

Possibly related discussion in #4345.

mmagnuski, 2017-06-27:
I don't have a strong opinion here - I'm not sure if it is necessary. I am usually annotating after filtering and if some operations can smooth the annoatated regions (for example tfr on raw) - I usually extend annotations adequately. 
With respect to discontinuities - eeglab for example marks them with specific events and does not cross these events when filtering. IIRC - all other markers/annotations are ignored.

Eric89GXL, 2017-06-27:
> I'm not sure if it is necessary. 

At least at the `concatenate_raw` boundaries I've seen some ugly artifacts that are big enough to break EOG detection (whereas filtering each raw before concat works fine).

> With respect to discontinuities - eeglab for example marks them with specific events and does not cross these events when filtering. IIRC - all other markers/annotations are ignored.

Okay we can do this by default. We can add some other annotation to our raw during concatenation (in addition to "BAD" maybe "EDGE") and filtering will respect it. Like `raw.filter(..., chunk='EDGE')` or you can do `raw.filter(..., chunk=('EDGE', 'BAD'))` to avoid both types.

mmagnuski, 2017-06-27:
I like that - adding 'EDGE' annotation automatically and the `chunk='EDGE'` option in filtering.

agramfort, 2017-06-27:
chunk -> ignore_annotation='EDGE' ok?

Eric89GXL, 2017-06-27:
Hmm... `ignore_annotation` isn't a great name because it's not clear if you're ignoring the annotation (what the var reads like), or using the annotation to choose sections to ignore (what we want to do). So maybe `annotation_ignore` or better `annotation_skip` so it reads a little better?

agramfort, 2017-06-27:
+1 for annotation_ignore

mmagnuski, 2017-06-27:
Maybe `ignore_by_annotation` (or `ignore_with_annotations`)? Not as easy to write but consistent with `reject_by_annotation`.

Eric89GXL, 2017-06-27:
I prefer `skip` to `ignore`, but I'm happy with either `skip_by_annotation` or `ignore_by_annotation`. I agree both are more consistent with `reject_by_annotation`.

mmagnuski, 2017-06-27:
I'm fine with both `skip_by_annotation` and `ignore_by_annotation` but like `skip_by_annotation` a little more.

Eric89GXL, 2017-06-27:
@agramfort `skip_by_annotation` okay with you?

agramfort, 2017-06-27:
ok !

kingjr, 2017-07-31:
Not sure if relevant comment, but the least surprise for me would be to filter concatenated raw files as if they were separate files. Artificially stitching raw segments as if they were continuous can always be done manually using mne.io.RawArray

Eric89GXL, 2017-07-31:
@kingjr that is what is proposed here (second paragraph in issue summary, plus continued discussion) so I'm glad you agree :)

Eric89GXL, 2017-07-31:
... and we should probably have this for 0.15. I should have time to tackle it in the next two weeks.

dengemann, 2017-07-31:
For me the question would be if naïvely filtering bad segments can spread
around noise and deteriorate the signal quality. Can we be sure that it is
not the case? I would intuitively suggest to checkout what is worse,
ignoring it or leaving bad segments out. The HCP data may be of help here
as they have many bad segments annotations.

Eric89GXL, 2017-07-31:
I think people can use `BAD` for any reason, including e.g. subject (on video) not paying attention, so I don't think it's good to skip them by default. I suspect this is also why EEGLAB doesn't pay attention to them by default, either. I'd rather cover this in documentation.

dengemann, 2017-07-31:
Yes this makes sense. What is BAD for the final analysis can be good enough
go be kept for filtering. It would be good to know how bad is too bad for
filtering. When should one not keep bad segments for filtering?

PR #4470 (2017-08-16): MRG+2: Make raw.filter annotation-aware
* Adds `skip_by_annotation` as discussed in #4353, with deprecation cycle to get people on board
* fixes a small bug with `Annotations` where the `BAD segment` wasn't actually being put in the correct spot
* changes `BAD segment` to have zero duration
* adds `EDGE segment` that is equivalent to `BAD segment`
* adds `__repr__` to `Annotations`

Ready for review/merge from my end.

Closes #4353.
closed 2017-08-16 by larsoner