mne-tools/mne-python #4360: Morphing one hemisphere to the other

christianbrodbeck, 2017-06-29:
I want to morph source-space data from one hemisphere to the other to test for lateralization differences. I found that [FreeSurfer includes a brain made for this purpose](https://surfer.nmr.mgh.harvard.edu/fswiki/Xhemi), with a cross-hemisphere surface registration. However from what I could find the mne-Python code currently does not allow using custom reg files. Would it make sense to refactor the code to make morph maps between arbitrary registration files and expose the functionality in `mne.compute_morph_matrix`?

christianbrodbeck, 2017-06-30:
> Is it an option instead to morph to this subject and compare there easily?

Unfortunately I don't think so, the way it is implemented is that the `fsaverage_sym` brain provides a surface registration for the left hemisphere which is registered to the right hemisphere of `fsaverage` and vice versa.

> I'd rather not have to expose the reg option if we can avoid it, but if we must I guess we can. What would the API look like to do what you want?

I was wondering whether there might be other use cases where it will be useful to specify two registration files directly, in which case I was thinking of adding to `compute_morph_matrix` an option like `from_name` and `to_name` to specify the filename (e.g. `compute_morph_matrix(... from_name="rh.spehere.reg", to_name="rh.sphere.left_right")`)

Another option if we think this is the only use case that will ever be relevant could be to use `compute_morph_matrix(... xhemi="rh")` and hard-code `rh.sphere.left_right`? But then if someone else creates a new brain on which to compare the hemispheres and they use a different name for the registration file then it already won't work anymore....

Eric89GXL, 2017-07-10:
I can live with new params. Maybe it could be `from_reg='sphere', to_reg='sphere'` default and you would use `to_reg='sphere.left_right'` for your use case? (The `lh.` and `rh.` would be concatenated internally as necessary.)

closed 2017-10-22 by christianbrodbeck