mne-tools/mne-python #4361: EDF events inconsistent?

cbrnr, 2017-06-30:
@jaeilepp recently fixed #4340 in #4341. Now I'm wondering why `mne.find_events` and `raw.get_edf_events` return different (numbers of) events. Shouldn't these methods yield consistent (i.e. identical) results? Sorry if you have already explained this before, but I still don't really understand it.

Consider [this test EDF+ file](http://www.teuniz.net/edf_bdf_testfiles/eeg_recording.zip). It consists of 37 EEG channels and 41 events. `raw.get_edf_events()` correctly returns a list with 41 elements, but `mne.find_events(raw)` returns only 18 events.

jaeilepp, 2017-06-30:
Output from ``get_edf_events``:
```
[[0.0, 0, u'REC START IIB CAL   '],
 [0.0, 0, u'PAT IIB EEG         '],
 [1.0, 0, u'A1+A2 OFF           '],
 [70.0, 0, u'od                  '],
 [121.0, 0, u'ademh bevestigen    '],
 [302.0, 0, u'ook OV5 (thermistor)'],
 [425.0, 0, u'oo                  '],
 [462.0, 0, u'od                  '],
 [464.0, 0, u'oo                  '],
 [560.0, 0, u'od                  '],
 [1127.0, 0, u'hoest               '],
 [1240.0, 0, u'aanspreken          '],
 [1262.0, 0, u'HVT START           '],
 [1292.0, 0, u'HVT 00:30           '],
 [1322.0, 0, u'HVT 01:00           '],
 [1352.0, 0, u'HVT 01:30           '],
 [1382.0, 0, u'HVT 02:00           '],
 [1412.0, 0, u'HVT 02:30           '],
 [1442.0, 0, u'HVT 03:00           '],
 [1450.0, 0, u'HVT END             '],
 [1480.0, 0, u'POST HVT 00:30      '],
 [1510.0, 0, u'POST HVT 01:00      '],
 [1592.0, 0, u'oo                  '],
 [1632.0, 0, u'od                  '],
 [1640.0, 0, u'PHOTO 10Hz          '],
 [1650.0, 0, u'PHOTO 12Hz          '],
 [1660.0, 0, u'PHOTO 14Hz          '],
 [1670.0, 0, u'PHOTO 15Hz          '],
 [1680.0, 0, u'PHOTO 16Hz          '],
 [1690.0, 0, u'PHOTO 17Hz          '],
 [1700.0, 0, u'PHOTO 18Hz          '],
 [1710.0, 0, u'PHOTO 19Hz          '],
 [1720.0, 0, u'PHOTO 20Hz          '],
 [1730.0, 0, u'PHOTO 1Hz           '],
 [1740.0, 0, u'PHOTO 3Hz           '],
 [1750.0, 0, u'PHOTO 5Hz           '],
 [1760.0, 0, u'PHOTO 7Hz           '],
 [1770.0, 0, u'PHOTO 9Hz           '],
 [1780.0, 0, u'PHOTO 11Hz          '],
 [1790.0, 0, u'PHOTO 13Hz          '],
 [1800.0, 0, u'PHOTO 50Hz          ']]

In [13]: raw.times[-1]
Out[13]: 1454.4760000000001
```

So the first two get lost since they're at sample zero (rising edge not detected). The last 21 are outside the data range. Maybe there could be a warning when such events are detected at read stage, but I'm afraid that would mean raising a warning almost every time an edf file with annotations is read.

cbrnr, 2017-06-30:
Something else must go wrong here, because this signal is supposed to be 363620 samples long. It looks like the sample frequency was not correctly detected. It should be 200Hz, but `raw.info["sfreq"]` says it's 250Hz.

jaeilepp, 2017-06-30:
Ok. It seems that the sfreq is defined to be max of all the sfreqs and there's one channel that has sfreq of 250. I'll make a PR.

cbrnr, 2017-06-30:
I didn't know that EDF files can have channels with multiple sampling frequencies. I guess the annotations will be associated with their own fs (200Hz in this case).

Which channel has a higher fs? How is this handled in the visualization? And in signal representation as an np.array? Are all channels upsampled to the highest fs?

jaeilepp, 2017-06-30:
The annotation channel has sfreq of 250 while the rest are at 200. It's not just the viz. The whole data gets messed when the sfreq is inferred wrong.

cbrnr, 2017-06-30:
The annotations are sampled at 250Hz in the EDF file? Or does this happen internally when the events are converted to a stim channel?

jaeilepp, 2017-06-30:
> The annotations are sampled at 250Hz in the EDF file

Yes.

cbrnr, 2017-06-30:
Which is not a problem because we just convert it to seconds, right?

jaeilepp, 2017-06-30:
Yeah.

cbrnr, 2017-06-30:
Cool. Do you want me to fix this or are you taking care of it?

jaeilepp, 2017-06-30:
see https://github.com/mne-tools/mne-python/pull/4363

closed 2017-07-09 by agramfort