mne-tools/mne-python #4368: convert_forward_solution

joewalter, 2017-07-03:
In mne.forward, there's a function called `_to_fixed_ori`. 
I expected that 
```
# use convert_forward_solution
forward_convert = convert_forward_solution(
    forward_ori, surf_ori=True, force_fixed=True, copy=True)
```
give the same result as
```
# use _to_fixed_ori
forward_fixed = convert_forward_solution(
    forward_ori, surf_ori=True, force_fixed=False, copy=True)
forward_fixed = _to_fixed_ori(forward_fixed)
```
Both expression should do the same, i.e., rotate to surf_ori and then convert to fixed.

The problem is that the results differ (`assert_allclose(forward_convert['sol']['data'], forward_fixed['sol']['data'])` fails).

So which way is recommended?
Should we get rid of one of the options (either by deleting `_to_fixed_ori`) or by calling `_to_fixed_ori` in `convert_forward_solution`?

agramfort, 2017-07-03:
you should use convert_forward_solution

it's a much more complete function.

joewalter, 2017-07-05:
The problem is that `convert_forward_solution` applies info on patch normals if `surf_ori=True`, but doesn't use patch normals if `force_fixed=True`. 

So if you do 
```
convert_forward_solution(forward_ori, surf_ori=True, force_fixed=False, copy=True)
forward_fixed = _to_fixed_ori(forward_fixed)
```
The normal orientation is based on patch normals. 

Otherwise
`forward_convert = convert_forward_solution(forward_ori, surf_ori=True, force_fixed=True, copy=True)`
the normal orientation is not based on the average.

This can be easily tested by forcing `use_ave_nn = False` convert_forward_solution. In this case both solutions are equal as shown by `assert_allclose(forward_convert['sol']['data'], forward_fixed['sol']['data'], rtol=1e-6)`

Wouldn't it be better to always use patch info if available to convert to surface orientation and then pick the corresponding parts of the solution matrix if `force_fixed = True` as done by `_to_fixed_ori`.

agramfort, 2017-07-05:
@mshamalainen or @Eric89GXL any idea why fixed the ori for historical MNE
can does not use patch info?

In other words this does not use patch info AFAIK:

forward_convert = convert_forward_solution(forward_ori, surf_ori=True,
force_fixed=True, copy=True)

PR #4620 (2017-09-28): FIX: Modified convert_forward_solution
Rebased version of #4372.

closes #4393, closes #4368
closed 2017-09-28 by agramfort