mne-tools/mne-python #4369: Neuroscan (cnt) - UnicodeDecodeError

gacek91, 2017-07-04:
Hey there,

I'm trying to read some Neuroscan *.cnt files with standard 10-20 montage (code below):
`raw = mne.io.read_raw_cnt(fls_path[-5],montage='standard_1020')`

Getting this error:

`<ipython-input-28-1e9f9ecff682>:1: RuntimeWarning:   Could not parse meas date from the header. Setting to [0, 0]...
  raw = mne.io.read_raw_cnt(fls_path[-5],montage='standard_1020')
/Users/Gacek/mne-python/mne/io/cnt/cnt.py:189: RuntimeWarning: overflow encountered in int_scalars
  data_size // (n_samples * n_channels) not in [2, 4]):
Could not define the number of bytes automatically. Defaulting to 2.
<ipython-input-28-1e9f9ecff682>:1: RuntimeWarning: Could not define the number of bytes automatically. Defaulting to 2.
  raw = mne.io.read_raw_cnt(fls_path[-5],montage='standard_1020')
Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/IPython/core/interactiveshell.py", line 2847, in run_code
    exec(code_obj, self.user_global_ns, self.user_ns)
  File "<ipython-input-28-1e9f9ecff682>", line 1, in <module>
    raw = mne.io.read_raw_cnt(fls_path[-5],montage='standard_1020')
  File "/Users/Gacek/mne-python/mne/io/cnt/cnt.py", line 95, in read_raw_cnt
    preload=preload, verbose=verbose)
  File "/Users/Gacek/mne-python/mne/io/cnt/cnt.py", line 362, in __init__
    data_format, date_format)
  File "/Users/Gacek/mne-python/mne/io/cnt/cnt.py", line 213, in _get_cnt_info
    ch_name = read_str(fid, 10)
  File "/Users/Gacek/mne-python/mne/io/utils.py", line 182, in read_str
    return str(bytestr.decode('ascii'))  # Return native str type for Py2/3
UnicodeDecodeError: 'ascii' codec can't decode byte 0xf6 in position 0: ordinal not in range(128)
`
Never had any such problems with MNE before so I'm clueless on how to solve this issue. Would be very glad for any suggestions.

Regards,

Mat Gaca

jaeilepp, 2017-07-04:
There's a number of things that can be wrong here. Looks like the header is not very intact (failing to read the number of bytes and measurement date). Would it be possible to share the file.

gacek91, 2017-07-04:
Thanks for a fast answer. Sure thing: [https://drive.google.com/file/d/0B6l1GJnvUSylWUhjYllvcmh1SXc/view?usp=sharing](url)

jaeilepp, 2017-07-04:
These are RIFF format, that are not supported at the moment.

jaeilepp, 2017-07-06:
FYI, there is an adapter for these files available [here](https://github.com/behinger/mne_tools), but since I can't find proper documentation for these files, I don't see the reader coming to mne anytime soon. Closing this.

closed 2017-07-06 by jaeilepp

smathot, 2019-10-25:
I also ran into this issue, and because of the uninformative error message it took me a moment to arrive at this page and find out that it's probably a file format issue. Would it be an idea to re-open this issue, and simply catch the `Exception` to translate into something less scary and more informative?

agramfort, 2019-10-25:
would you consider sending us a pull request?

>

hernandezurbina, 2022-08-08:
hi all,

I'm new to the world of EEG data analysis. My team generated CNT files from ANT neuro and I am trying to load them using read_raw_cnt from python mne. I run into the same issue as described above:

`'ascii' codec can't decode byte 0x93 in position 1: ordinal not in range(128)`

What could be causing the problem? I opened the first 100 lines of the file and saw a bunch of strange characters. Thanks!

cbrnr, 2022-08-08:
We do not support importing ANT CNT files, please see #3609.

hernandezurbina, 2022-08-08:
@cbrnr got it! Thanks for the quick reply!

romybeaute, 2023-05-12:
Hello everyone, did someone find a solution for this? I am having the same problem, with ANT CNT ...

drammock, 2023-05-12:
@romybeaute see @cbrnr's comment two lines up from yours.  You don't need to read the entire thread of #3609, just skip to this comment: https://github.com/mne-tools/mne-python/issues/3609#issuecomment-1272557341 and read on from there.  It explains 3 possible work-arounds.