mne-tools/mne-python #4370: Option to disable interactive selection in raw.plot_psd(average=False)

cbrnr, 2017-07-04:
When plotting a PSD for all channels (i.e. `raw.plot_psd(average=False)`, it is possible to select a portion of the data by clicking and dragging with the mouse. However, this results in an error if no topographic information is available. I'd vote for a new argument to turn this interactive selection off in this method and all subsequent functions - thoughts?

mmagnuski, 2017-07-04:
Why would you want to click the psd figure other than to get the topomap? In general I think adding `interactive` argument is ok but it would be most useful in functions that with interactive mode add buttons / sliders which are unnecessary if you want just viz without interactivity.

cbrnr, 2017-07-04:
PSD plots show the channel labels if you click on a trace - this is very convenient and I'd like to keep this feature. However, it is easy to accidentally drag the mouse, which leads to an error. It would also be helpful if this topomap functionality (i.e. clicking and dragging) was automatically disabled if there is no topographic information available.

agramfort, 2017-07-04:
+1 to automatically disable and put a warning in the console

mmagnuski, 2017-07-04:
@cbrnr Right, so I'm +1 too. What happens if you have channel positions and try to click on a channel - does the label show up or a topomap?

mmagnuski, 2017-07-04:
Ah, ok I read that you said that it is easy to drag the mouse by accident.

cbrnr, 2017-07-04:
The label shows up. The topomap is only plotted when clicking and dragging a rectangle.

closed 2017-07-06 by agramfort