mne-tools/mne-python #4374: ICA mixing matrix changes when saved and reloaded

jpazdera, 2017-07-06:
I noticed that if you run ICA with the ICA fit() method, then save and reload the ICA object with read_ica(), the loaded object ends up with a different mixing matrix from the original object. In other words, if you do the following:

```
# Assume raw is a Raw object and path is the location where we wish to save our ICA solution
ica = ICA(method='fastica')
ica.fit(raw)
ica.save(path)
ica2 = mne.preprocessing.read_ica(path)
```

Then ica.mixing_matrix_ and ica2.mixing_matrix_ are not the same matrix. For the data I'm currently working with, the first and last row of the two matrices are approximately identical, but all other rows are quite different.

After looking into the code for fitting and loading ICA, I think this is because of the following: When you run ica.fit(), the mixing matrix is calculated as linalg.pinv(unmixing_matrix). Afterwards, the columns of the mixing matrix and the rows of the unmixing matrix are sorted by the explained variance of their corresponding component. When you later load the ICA object from a file, the mixing matrix is once again calculated (rather than directly loaded from the save file) as linalg.pinv(unmixing_matrix); however, because the saved unmixing matrix has already been sorted, this results in a different mixing matrix from when we originally fit the ICA.

Now my linear algebra isn't the greatest, so I'm not certain whether this is actually an issue. I just figured I'd bring it up!

(P.S. Thanks for solving the [issue](https://github.com/mne-tools/mne-python/issues/4028) with reading .bdf files where num_records == -1. I hadn't checked on that thread in months, but just recently saw that it's been fixed in v0.15!)

agramfort, 2017-07-07:
indeed it seems like a bug...

can you give a code snippet with mne sample data to replicate the pb?

jpazdera, 2017-07-07:
I tried running ICA on the mne sample dataset with the code below, and it did not reproduce the issue. I'm also having trouble reproducing it after re-running ICA on one of the EEG sessions I was having trouble with yesterday. Furthermore, I checked it with a friend of mine, who is a grad student in mathematics, and the two methods of calculating the mixing matrix here (finding the inverse, then sorting vs sorting, then finding the inverse) _should_ be equivalent.

It's possible that something else is going on, but for now I'll just keep an eye on it and perhaps save the original mixing matrix just in case.

```
import mne
import numpy as np

# Load sample data and run ICA on it, then save
path = mne.datasets.sample.data_path()+'/MEG/sample/sample_audvis_filt-0-40_raw.fif'
raw = mne.io.read_raw_fif(path)
ica = mne.preprocessing.ICA(method='fastica')
ica.fit(raw)
ica.save('test-ica.fif')

# Load the saved ICA object and check for differences in the mixing matrices
ica2 = mne.preprocessing.read_ica('test-ica.fif')
print 'Differences: ', np.sum(np.logical_not(np.isclose(ica.mixing_matrix_, ica2.mixing_matrix_)))
```

agramfort, 2017-07-08:
is it possible that we broke backward compat with old ICA files when
introducing the sorting?

mmagnuski, 2017-07-08:
If the matrices are just inverted upon reading - it shouldn't happen.
@jpazdera - are you sure that the first time you experienced this problem you didn't compare weights from two ICA runs?

jpazdera, 2017-07-10:
I don't believe it's a backwards compatibility issue, as I think I've only used v0.15 in my testing. I also don't think I was comparing weights from two ICA runs, as the unmixing matrices were identical between the fit and loaded versions of the ICA object - just the mixing matrices were different. 

I haven't been able to reproduce the issue, though, so I'd say just ignore it for now and I'll keep an eye out in case it happens again.

Eric89GXL, 2017-07-10:
Okay closing for now, let us know if you can reproduce

closed 2017-07-10 by Eric89GXL