mne-tools/mne-python #4375: Volume_Source_Space question

bloyl, 2017-07-07:
I'm trying to setup a sparse (10-15 sources) volume source space, similar to the standard source spaces used in besa. I'm using setup_volume_source_space but I'm unclear what correct coordinate frame I should use for pos['rr'].

I started by using mri ras coordinates. However when computing fwd models some sources were being excluded for being outside of the BEM inner skull. In trying to track this down I found the following

https://github.com/mne-tools/mne-python/blob/master/mne/viz/misc.py#L306-L307

Which suggests that the voxel to MNE mri RAS transform is a bit strange. I assume that this has to do with the coordinate frame used by freesurfer for the surface files? Is this modified transform used consistently through out MNE to map coordinates to MR locations?

Should I be using this modified transform to compute pos['rr'] from RAS MRI world coordinates (such as those available from fslview)?

Thanks.
Luke

agramfort, 2017-07-08:
what you want is what we call a "discrete source space".

You need to call setup_volume_source_space with pos being a dict as here:

https://github.com/mne-tools/mne-python/blob/master/mne/tests/test_source_space.py#L190

unfortunately it does not seem to be illustrated with an example.

let me know if you need more info

bloyl, 2017-07-10:
Thanks Alex,

The issue is that from the documentation for setup_volume_source_space its unclear what the coordinate frame should be for the locations that you pass in.

I did what I suspect many users would do which is to open the T1.mgz in a viewer (Slicer3d or paraview etc) and then use the RAS locations returned from the viewer. This doesn't work as it seems like internal MRI coordinate frame used by freesurfer (and MNE) doesn't respect the origin of the voxel to ras transform. 

When I use the modified vox2ras transform to map the images RAS coordinates into the modified freesurfer space, I get much more reasonable results and the mne-python visualization shows the sources in the correct anatomy.

If there is interest and i have some time I'll put together a PR to simplify the coordinate transforms and illustrate the issue.

Eric89GXL, 2017-07-10:
> When I use the modified vox2ras transform to map the images RAS coordinates into the modified freesurfer space, I get much more reasonable results and the mne-python visualization shows the sources in the correct anatomy.

Perhaps this is what comes up as `vox2ras_tkr` in `nibabel`?

> If there is interest and i have some time I'll put together a PR to simplify the coordinate transforms and illustrate the issue.

Yes please, we definitely need some coordinate frame tutorial to talk about these issues. If it's comprehensive that's awesome, but even something simple to start that we can build from would help.

bloyl, 2017-07-10:
I had forgotten about `vox2ras_tkr`. at least in my test subject vox2ras_tkr matches the "hack" from https://github.com/mne-tools/mne-python/blob/master/mne/viz/misc.py#L306-L307

Is it safe to assume that the tkr (tkregsiter) https://surfer.nmr.mgh.harvard.edu/fswiki/CoordinateSystems is in fact the coordinate frame that is used by the BEM surfaces and is therefor the MNE-MRI coordinate frame?

Eric89GXL, 2017-07-10:
Yes I think it should be the case / the correct thing to use.

larsoner, 2019-08-20:
Nothing to do here other than improve our docs, probably having to do with coordinate frames, which we have noted elsewhere. So closing

closed 2019-08-20 by larsoner