mne-tools/mne-python #4377: MEG Interpolation

Chrisinger, 2017-07-10:
Dear MNE-community,

I am currently trying to interpolate some bad MEG channels with MNE.
Unfortunately there seems to occur an exception, such that the resulting
channels hold only zeros. Even when I try to interpolate only one
channel I get from '_field_interpolation.py:80' the log
'Truncate at 0 missing 0.0001'
As I understand it, this happens because the eigenvalues are all too
small. My data contains of 248 channels measured at ~291000 time steps.

I am currently stuck there, with no idea remaining. Can you give me a
hint what else I could try to get at least some interpolation for my bad
channels?

As suggested, I attached a minimal example.

[interpol.zip](https://github.com/mne-tools/mne-python/files/1136430/interpol.zip)

Thanks in advance, Christoph

agramfort, 2017-07-22:
@Eric89GXL are you on it?

Eric89GXL, 2017-07-24:
This example runs with no problem on my machine. Can you try updating to latest MNE `master` to see if it fixes the problem?

If it doesn't, what is your `mne.sys_info()`?

Eric89GXL, 2017-07-24:
... oh wait, the channel is all zero. I'll see if I can make sense of it.

PR #4418 (2017-08-17): MRG: Fix interpolation truncation
I'm not actually sure if we want to do +1 here (which means "missing less than 0.0001") or do `max(old_val, 1)`, which means "missing at least 0.0001 but keeping at least one component". The former seemed nicer to me but is a bit bigger of a change.

Closes #4377.
closed 2017-08-17 by larsoner