mne-tools/mne-python #4381: drop_channels without preloading

skjerns, 2017-07-14:
I remember in a previous version I was able to drop channels without preloading the data.

Now this is not possible anymore.

What would be the most simple (and fastest) way to get all data from a certain selection of channels as a numpy array?

Right now I'm loading all data with `header.load_data()` and then I extract the channels with `header.get_data([0,1,2])`. I fear that somewhere on the way there might be a memory leakage (I had this before), as my memory consumption is far above the final numpy array.

agramfort, 2017-07-14:
no need to preload to get the data of a few channels

raw = read_raw_xxx(..., preload=False)
data, times = raw[picks, :]

skjerns, 2017-07-14:
great, thanks! works perfectly. Who would have thought that's so easy (well probably it was at an obvious point in the manual that I didn't see).

Eric89GXL, 2017-07-14:
> well probably it was at an obvious point in the manual that I didn't see

I'd rather think of it as, if you looked and didn't see it, it wasn't obvious enough. Where did you look? Any time to make a PR to add a line to help the next person?

skjerns, 2017-07-18:
I expected to find that at http://martinos.org/mne/stable/manual/memory.html

But also generally at http://martinos.org/mne/stable/manual/#reading-your-data

I'll try to add that to the docs in the next days.

agramfort, 2017-07-18:
> I'll try to add that to the docs in the next days.

thanks a lot.

jaeilepp, 2017-08-01:
I believe this is solved.

closed 2017-08-01 by jaeilepp