mne-tools/mne-python #4385: issues in test_docstring_parameters

amueller, 2017-07-15:
The way that walk_packages is called in the docstring tests that got ported to scikit-learn means that the tests only work when called from a specific location. Also it misses all subpackages.
The right call should look like 
```python
                      for pckg in walk_packages(prefix='mne.',                                                                
                                                path=mne.__path__) 
```

Eric89GXL, 2017-07-17:
We currently call:
```
for importer, modname, ispkg in walk_packages(mne.__path__, prefix='mne.'):
```
And I think you're just suggesting to change the `arg` to a `kwarg` as `path=mne.__path__`, but [path is the first argument](https://docs.python.org/3.6/library/pkgutil.html#pkgutil.walk_packages) so this shouldn't matter. I must be missing something...?

amueller, 2017-07-17:
Huh, ok. I didn't look at the code, but the code @agramfort copied to sklearn looked very different. Your seems to be fine.

closed 2017-07-17 by amueller

Eric89GXL, 2017-07-17:
He probably copied over an old version long ago. Thanks for checking back in any case