mne-tools/mne-python #4386: Generate a button to see changed documentation

choldgraf, 2017-07-15:
I just found out about an awesome little feature that the sklearn devs built for their documentation. 

They've configured their circle file to generate a folder with a list of files that (they think) have changed, and puts it in a folder.

https://github.com/scikit-learn/scikit-learn/blob/master/build_tools/circle/build_doc.sh#L134

and they have a tampermonkey/greasemonkey script that looks for that folder and generates a button that gives you a list of links to the changed HTML pages in the docs:

https://gist.github.com/lesteve/470170f288884ec052bcf4bc4ffe958a

It seems like this wouldn't be too tough to implement in MNE since we're already building with circle...could be a nifty feature to have for documentation improvements. WDYT @Eric89GXL ?

Eric89GXL, 2017-07-17:
+1 would be cool to have

I'm not motivated to do it currently, but maybe someday :)

larsoner, 2019-08-20:
Closing as YAGNI but if someone is motivated feel free

closed 2019-08-20 by larsoner