mne-tools/mne-python #4390: Report example failing

kambysese, 2017-07-17:
I was under the impression `report.parse` can generate reports if e.g., anatomy files are missing until I just hit an `IOError` from `_render_bem` concerning missing `T1.mgz`. Trying to debug now with example dataset and gist from [here](http://martinos.org/mne/stable/tutorials/report.html?highlight=mne%20report) is failing. Any ideas what this might be?

```
import os
assert(os.environ['SUBJECTS_DIR'] == '/media/ktavabi/ALAYA/data/freesurfer')
from mne.datasets import sample
from mne.report import Report
path = sample.data_path()
report = Report(verbose=True)
report.parse_folder(data_path=path, pattern=['*audvis_raw.fif', '*-eve.fif'])
```
Throws back ...

> <ipython-input-1-277e9b18892d>:13: RuntimeWarning: `info_fname` not provided. Cannot render -cov.fif(.gz) and -trans.fif(.gz) files.
>   report.parse_folder(data_path=path, pattern=['*audvis_raw.fif', '*-eve.fif'])
> `info_fname` not provided. Cannot render -cov.fif(.gz) and -trans.fif(.gz) files.
> Iterating over 6 potential files (this may take some time)
> Rendering : /home/ktavabi/Projects/mne-python/examples/MNE-sample-data/MEG/sample/sample_audvis_raw.fif
> Opening raw data file /home/ktavabi/Projects/mne-python/examples/MNE-sample-data/MEG/sample/sample_audvis_raw.fif...
>     Read a total of 3 projection items:
>         PCA-v1 (1 x 102)  idle
>         PCA-v2 (1 x 102)  idle
>         PCA-v3 (1 x 102)  idle
>     Range : 25800 ... 192599 =     42.956 ...   320.670 secs
> Ready.
> Current compensation grade : 0
> Rendering : /home/ktavabi/Projects/mne-python/examples/MNE-sample-data/MEG/sample/sample_audvis_eog-eve.fif
> 
> ** (python:24079): WARNING **: Error retrieving accessibility bus address: org.a11y.Bus.Error: Failed to execute child process "/bin/dbus-daemon" (No such file or directory)
> Rendering : /home/ktavabi/Projects/mne-python/examples/MNE-sample-data/MEG/sample/sample_audvis_ecg-eve.fif
> Rendering : /home/ktavabi/Projects/mne-python/examples/MNE-sample-data/MEG/sample/sample_audvis_filt-0-40_raw-eve.fif
> Rendering : /home/ktavabi/Projects/mne-python/examples/MNE-sample-data/MEG/sample/sample_audvis_raw-eve.fif
> Rendering : /home/ktavabi/Projects/mne-python/examples/MNE-sample-data/MEG/sample/ernoise_raw-eve.fif
> <ipython-input-1-277e9b18892d>:13: RuntimeWarning: `subjects_dir` and `subject` not provided. Cannot render MRI and -trans.fif(.gz) files.
>   report.parse_folder(data_path=path, pattern=['*audvis_raw.fif', '*-eve.fif'])
> `subjects_dir` and `subject` not provided. Cannot render MRI and -trans.fif(.gz) files.

agramfort, 2017-07-17:
import os
from mne.datasets import sample
from mne.report import Report
path = sample.data_path()
subjects_dir = os.path.join(path, 'subjects')
subject = "sample"
report = Report(verbose=True, subjects_dir=subjects_dir, subject=subject)
report.parse_folder(data_path=path, pattern=['*audvis_raw.fif',
'*-eve.fif'])

should work. Do you confirm?

kambysese, 2017-07-17:
Confirm Thanks. I'll close. Figured out what other issue was.

closed 2017-07-17 by kambysese