mne-tools/mne-python #4391: Time-frequency of a single-channel signal

DominiqueMakowski, 2017-07-18:
Hey, I'm trying to extract the power, over time, of certain frequency bands of a unique (single-channel) signal sampled at 100Hz. However, while I know that some of these bands should be absent (e.g., the ULF (ultra low frequency) because the signal is too short for withholding variations within it), the results are rather odd: all time-frequencies powers present the same pattern than the original signal... Here's a reproducible code snippet:

```python
import numpy as np
import pandas as pd
import mne

signal = pd.read_csv("signal_100Hz.txt")["Signal"]
sampling_rate=100

# Set frequencies with variable step to have approximately the same amount of values in each band
tf = {"ULF": np.arange(0.0001, 0.0033, 0.001),
      "VLF": np.arange(0.0033, 0.045, 0.005),
      "LF": np.arange(0.04, 0.16, 0.01),
      "HF": np.arange(0.15, 0.42, 0.02),
      "VHF": np.arange(0.4, 0.51, 0.01)}

for freqs_range in tf:
    freqs = tf[freqs_range]
    signal = mne.time_frequency.tfr_array_multitaper(np.array([[signal]]), sampling_rate, freqs, n_cycles=freqs/2, zero_mean=False, time_bandwidth=7)[0][0]
    signal = np.mean(signal, 0)  # Average
    tf[freqs_range] = signal  # Replace data in dict

tf = pd.DataFrame.from_dict(tf)
tf["Raw_Signal"] = signal
tf.plot()
```
[image]

Is that correct? Data is attached here: [signal_100Hz.txt](https://github.com/mne-tools/mne-python/files/1155076/signal_100Hz.txt)

Thanks!!

agramfort, 2017-07-18:
you should use band pass filtering for this.

raw.filter(l_freq, h_freq)

for each band.

then you can use for example hilbert to get power over time.

it's weird to use TFR with uncontrolled frequency resolution due to tapering
to just have a temporal signal.

mmagnuski, 2017-07-18:
Another thing to note is that all your frequencies are low (`< 1 Hz`) and so when you are setting `n_cycles` to `freqs / 2` (0.5 second) you end up using `n_cycles < 0.5` which is not enough to get meaningful frequency resolution - so you end up replicating the time domain data. Try not to have less than 3 `n_cycles` - but it is usually recommended to be at or above 5 or even 7 (as in the mne docs).

DominiqueMakowski, 2017-07-18:
You were right, filtering and then using hilbert was the right way to go. Sorry for the dumb question and  thanks a lot!

closed 2017-07-18 by DominiqueMakowski