mne-tools/mne-python #4396: Error when loading fif:  unsupported operand type(s) for -: 'NoneType' and 'NoneType'

bdyetton, 2017-07-18:
Hi,
When reading in a fif file that I had previously saved with mne I am getting this error:

  File "C:\Users\bdyet\Anaconda3\envs\pyeeg\lib\site-packages\mne\io\fiff\raw.py", line 177, in _read_raw_file
    duration = tag.data - onset
TypeError: unsupported operand type(s) for -: 'NoneType' and 'NoneType'

```
elif kind == FIFF.FIFF_MNE_BASELINE_MAX:
      duration = tag.data - onset
```

fif file is here:
[PAI_wd_sub2019_raw.zip](https://github.com/mne-tools/mne-python/files/1157586/PAI_wd_sub2019_raw.zip)

Is this a bug or something in the way i have saved the fif file?
Thanks

jaeilepp, 2017-07-19:
Looks like this was fixed in 0.14. Can you try updating MNE-python?

bdyetton, 2017-07-19:
Hmm, I am on v0.14 already. Should I update to  0.15 dev?

agramfort, 2017-07-19:
>
> Hmm, I am on v0.14 already. Should I update to 0.15 dev?
>

I am afraid so. We should release late August.

bdyetton, 2017-07-19:
Ok thanks for you help, ill update.

-- 
Regards,
*Ben Yetton*

MA Cog Psych,
BSc (Robotics) w. first class honors,
Graduate Student,
Mednick Sleep and Cognition Lab,
University Of California, Riverside
900 University Ave
Riverside, CA 92521
benjamin.yetton@email.ucr.edu

bdyetton, 2017-09-21:
Ok so after upgrading to 0.15.dev0 i am still getting the above error. It only occurs during loading of a fif that contains data that has been ica'ed.

bdyetton, 2017-09-22:
Also, are there limits on the length of each string that makes up a ch_name? When i import my ch_names seem to be clipped.

agramfort, 2017-09-22:
I just tried to read your file and I got no problem with current master:

```
In [2]: raw =  mne.io.read_raw_fif('PAI_wd_sub2019_raw.fif')

In [3]: raw.ch_names
Out[3]:
['F7',
 'Fz',
 'F5',
 'F8',
 'C3',
 'Cz',
 'P7',
 'P3',
 'Pz',
 'P4',
 'P8',
 'Oz',
 'events']
```

bdyetton, 2017-09-22:
Hmm, you're right. It appears that the issue occurs when I have just written a file and try to load it again right away.

If the names are: ```P3_20Hz-60Hz_Env_Filt``` then they get clipped to something like ```P3_20Hz-60Hz_En```
I will close this issue for now, but you may want to add that the ch_names have a max length when ```save```ing or fix the clipping in the ```read_raw_fif``` (I'm not sure where it occurs).

closed 2017-09-22 by bdyetton

larsoner, 2017-09-23:
I've retitled the issue to make it about channel names and will reopen it. It is probably a limitation of the FIF format that we need to document, but we'll need to check.

agramfort, 2017-09-24:
with this https://github.com/mne-tools/mne-python/pull/4600
the problem should not happen again.