mne-tools/mne-python #4397: Error when processing ICA:  ValueError: Input contains NaN, infinity or a value too large

bdyetton, 2017-07-18:
Hi,
Applying ICA to the following file and getting this error:

  File "C:\Users\bdyet\Anaconda3\envs\pyeeg\lib\site-packages\sklearn\utils\validation.py", line 58, in _assert_all_finite
    " or a value too large for %r." % X.dtype)
ValueError: Input contains NaN, infinity or a value too large for dtype('float64').

The error originates from this line in mne\preprocessing\ica.py:
`data = pca.fit_transform(data.T)`

One of my data channels is all infinities, however, it is not inf before running the ICA function.
Although I'm not sure if you can open this file because it has load errors, see #4396 

[PAI_wd_sub2019_raw.zip](https://github.com/mne-tools/mne-python/files/1157591/PAI_wd_sub2019_raw.zip)

Thanks!

jaeilepp, 2017-07-19:
I'm not sure how exactly you got the error, but fitting the ICA should not tamper with the raw data. Can you see if updating the version works here as well? If not, provide a small snippet of code to reproduce the error.

bdyetton, 2017-07-19:
Ok so i found where the error is coming from. There is a channel marked as bad, and that channels data get s "_pre_whiten"'ed until all its values are either + or - infinity.  Removing this channel (via picks) fixes the issue. Given that this channel is marked as bad, and generally mne ignores bad channels, i would consider this a bug (or at least it should be documented).  Furthermore, I think the "picks" attribute of the ICA.fit function should take channel names not channel indexes, this would match other functions in mne.

agramfort, 2017-07-19:
>
> Ok so i found where the error is coming from. There is a channel marked as
> bad, and that channels data get s "_pre_whiten"'ed until all its values are
> either + or - infinity.
>
can you provide a full gist to replicate? yes bad channel should be
ignored. You should not have to remove a bad channel.

> Furthermore, I think the "picks" attribute of the ICA.fit function should
> take channel names not channel indexes, this would match other functions in
> mne.
>

the semantic of picks is also indices. Otherwise it's called ch_names or
include.

bdyetton, 2017-07-19:
> the semantic of picks is also indices. Otherwise it's called ch_names or
> include.
Oh right, I guess I use picks less than I thought it did.

Ok with some further investigation, ICA.fit() will only try to process bad channels when the "picks" argument is supplied that includes the bad channel. I guess that could be expected behavior given that I have explicitly asked it to pick the bad eeg channel, however, i assumed that once a channel is marked as bad, it will never be processed. 
Maybe a warning could be added to say that you have asked for analysis of a bad channel?

Gist (will file attached above):
```
    import mne
    import matplotlib.pyplot as plt
    eeg = mne.io.read_raw_fif('PAI_wd_sub2019_raw.fif', verbose=True)
    eeg.load_data()
    eeg.info['bads'] = ['C3']
    ica = ICA()
    channels = [chan for chan in eeg.ch_names if chan != 'events']
    chan_idxs = [idx for idx, chan in enumerate(eeg.ch_names) if chan in channels]
    reject = dict(eeg=40e-5)

    #working
    ica = ica.fit(eeg, reject=reject)

    #Failing
    ica = ica.fit(eeg, picks=chan_idxs, reject=reject)
```

agramfort, 2017-07-19:
use the picks functions such as pick_types. They remove bad channels
automatically.

so for me there is no bug.

bdyetton, 2017-07-19:
Ok will do, Maybe a comment in the documentation under the "picks" argument would be helpful. Something like, "If indexes that represent bad channels are included, they will be analyzed"
Thanks for your help.

mmagnuski, 2017-07-19:
Reminding users that `picks` can override bad channel exclusion is good idea IMO.

agramfort, 2017-07-20:
no objection

PR more than welcome.

thx

commit 245d455b (2020-11-24): closes #4397 [skip azp][skip github] (#8567)
closed 2020-11-24 by larsoner