mne-tools/mne-python #4403: Defect montages shipped with MNE?

cbrnr, 2017-07-19:
As mentioned in #4400, these two montages seem to have duplicate channel positions:

- `mne/channels/montages/standard_postfixed.elc`
- `mne/channels/montages/standard_primed.elc`

Furthermore, we currently have no description for `mne/channels/montages/10-5-System_Mastoids_EGI129.csd`, which contains a total of 465 locations. The last 129 locations are just numbers. Besides the cumbersome file name, we should make sure that this is a valid and useful montage (this file was also not documented in the docstring of the `read_montage` function).

Eric89GXL, 2017-07-19:
@dengemann any insight on these montages? You added them ages ago (2014)

cbrnr, 2017-07-20:
Something else I've noticed. The elements in the `ch_names` list have different types, especially the ones with byte elements could be problematic:

```
10-5-System_Mastoids_EGI129 <class 'numpy.bytes_'>
biosemi128 <class 'numpy.str_'>
biosemi16 <class 'numpy.str_'>
biosemi160 <class 'numpy.str_'>
biosemi256 <class 'numpy.str_'>
biosemi32 <class 'numpy.str_'>
biosemi64 <class 'numpy.str_'>
easycap-M1 <class 'numpy.str_'>
easycap-M10 <class 'numpy.str_'>
EGI_256 <class 'numpy.bytes_'>
GSN-HydroCel-128 <class 'str'>
GSN-HydroCel-129 <class 'str'>
GSN-HydroCel-256 <class 'str'>
GSN-HydroCel-257 <class 'str'>
GSN-HydroCel-32 <class 'str'>
GSN-HydroCel-64_1.0 <class 'str'>
GSN-HydroCel-65_1.0 <class 'str'>
standard_1005 <class 'str'>
standard_1020 <class 'str'>
standard_alphabetic <class 'str'>
standard_postfixed <class 'str'>
standard_prefixed <class 'str'>
standard_primed <class 'str'>
```

Eric89GXL, 2017-07-20:
Yes we should also fix those

agramfort, 2017-07-28:
not fixed by #4427 ?

cbrnr, 2017-07-28:
Not entirely, there's still the issue with duplicate locations and the weird montage.

jaeilepp, 2017-07-28:
I guess the things in the original post are still unanswered.

Eric89GXL, 2017-08-05:
@dengemann any time to look in the next week or two to see if you can make sense of these files you added a few years back? It would be good to fix this for 0.15

larsoner, 2017-08-16:
ping @dengemann any insight on these?

dengemann, 2017-08-16:
I think we can remove those. Early attempts to support csd transform that
got stuck.

larsoner, 2017-08-16:
So remove both or just one of these:

* `mne/channels/montages/standard_postfixed.elc`
* `mne/channels/montages/standard_primed.elc`

dengemann, 2017-08-16:
Ah wait these I don't know. I think they must be from fieldtrip.

larsoner, 2017-08-16:
The mysterious `10-5-System_Mastoids_EGI129.csd` file is already gone from our repo, so I'll close this then

cbrnr, 2017-08-17:
@larsoner but what about the other two? We should either correct the duplicate locations or remove the montages (because we have other standard montages).

dengemann, 2017-08-17:
I am almost certain they are from Fieldtrip. Maybe we can see if they are
still the same there and pull their updates in case they received
corrections.

cbrnr, 2017-08-17:
Right, it looks like we use all Fieldtrip montages in MNE 😄 : https://github.com/fieldtrip/fieldtrip/tree/master/template/electrode

They haven't been updated in years, so they're identical to what we're using.

cbrnr, 2017-08-17:
You think the identical locations really have two alternative names? Even if this is true I'd still find it ugly, because then why bother with having four different montages for the standard locations when we could put all names into one?

larsoner, 2017-08-17:
> You think the identical locations really have two alternative names?

Look at the files, the channel names they contain are different

> why bother with having four different montages for the standard locations when we could put all names into one?

This would require re-working the reading functions, and probably wouldn't otherwise change the user-facing API, so I don't see much point

cbrnr, 2017-08-18:
> Look at the files, the channel names they contain are different

I actually looked at the montages from FieldTrip, and they look weird. Not only do some of them have channels with identical locations, but they look wrong in general. For example, the Cx channels should be the central channels from left to right ears - they are too posterior.

Compare one of the FieldTrip montages:
[image]

And here's a 10-20 montage from EEGLAB:
[image]

The Biosemi montages shipped with MNE look correct (i.e. just like the one from EEGLAB).

larsoner, 2017-08-18:
This looks like a coordinate frame problem -- the `FieldTrip` one looks suspiciously like it's not in head coords (a tilt forward of the head would make it look at least closer to the other one). Can you check to see if they are in head coords? (You can tell quickly because they RPA, LPA, and Nasion should each have two zero-valued coordinates.)

If that doesn't make them equivalent, then we might want to bother the FieldTrip folks to sort this out.

cbrnr, 2017-08-18:
Here are the coordinates for LPA, RPA, and Nz:

```
[ -8.60761000e-02,  -1.99897000e-02,  -4.79860000e-02]
[  8.57939000e-02,  -2.00093000e-02,  -4.80310000e-02]
[  8.30000000e-06,   8.68110000e-02,  -3.99830000e-02]
```

larsoner, 2017-08-18:
Yeah those are incorrect (not Neuromag head coords). Next up would be to try `get_ras_to_neuromag_trans` to get the transformation, and `mne.transforms.apply_trans` to the points (double-checking that LPA/Nasion/RPA are now in reasonable locations), setting those as `montage.pos`, and then re-plotting the positions. Feel free to do it or I can.

In any case, we'll almost certainly want to have our montages in head coords, so we should probably fix it.

larsoner, 2017-08-18:
Oh actually we have it already. You just need `transform=True` in `read_montage`. Can you try it to see if it fixes it?

larsoner, 2017-08-18:
@teonbrooks any idea why we default to `transform=False` in our montage-reading code? Everything else in MNE is in Neuromag head coords, seems like we'd want to (at least by default) make them in head coords, too.

cbrnr, 2017-08-18:
@larsoner yeah, this looks better:

[image]

```
array([[ -8.60701216e-02,   3.46944695e-18,   6.93889390e-18],
       [  8.57998854e-02,   0.00000000e+00,   0.00000000e+00],
       [ -2.40049137e-18,   1.07111604e-01,   6.93889390e-18]])
```

So are all montages shipped with MNE in other coordinates? In other words, should we always set `transform=True`? I noticed that it doesn't make a huge difference for e.g. `biosemi64` other than the transformed locations are a bit more condensed toward the center.

cbrnr, 2017-08-18:
Still I don't like the two montages with duplicate channel names...

larsoner, 2017-08-18:
> In other words, should we always set transform=True? 

Yes I think it's worth a deprecation cycle to go to `transform=True`, we assume EEG coords are Neuromag head coords in various places.

> Still I don't like the two montages with duplicate channel names...

You mean montages with duplicate positions and different channel names, right? What is your proposed alternative? Let's say we have two people, one whose channel names match those of `standard_postfixed.elc` and another whose match those of `standard_primed.elc`. Currently each just does `montage = read_montage(...)` each using a different name, and everything else works.

How do you propose this should change from the user-facing API?

cbrnr, 2017-08-19:
> Yes I think it's worth a deprecation cycle to go to transform=True, we assume EEG coords are Neuromag head coords in various places.

OK, but maybe it's `False` by default because it fails if the three fiducials are not in the montage? Why don't we ship all montages in head coordinates? Then `transform` could remain `False` for custom montage files.

> You mean montages with duplicate positions and different channel names, right? What is your proposed alternative? Let's say we have two people, one whose channel names match those of [...]

This is not quite what I meant. Currently we have overlapping channel names within one montage. I would just remove all duplicate names - these are all non-standard names anyway, and it is not that hard to rename channels in a raw file so that they match names in a given montage.

larsoner, 2017-08-19:
OK, but maybe it's False by default because it fails if the three fiducials
are not in the montage? Why don't we ship all montages in head coordinates?
Then transform could remain False for custom montage files.

This would make it harder to copy paste files between packages and doesn't
buy us much over using transform.

One thing to note is that, for sensor positions defined on a sphere (with
fiducials that are or should also be on that sphere), knowing what the head
coordinate frame is ends up being equivalent to knowing the locations of
the three fiducials. So in order to get to your proposal (ship all in head
frame) we essentially need to know the fiducials anyway. Thus as soon as we
make it possible to get to head coordinates, we obviate the need to
actually do the transformation to the points saved on disc.

And for the files that don't have the fiducials, the diff to other packages
or standard montage files is still smaller by just adding the fiducials. So
it still makes maintenance and interoperability easier.

This is not quite what I meant. Currently we have overlapping channel names
within one montage.

I somehow missed this and it's not easy to check ATM (on mobile), do each
of the two listed in the top comment have this problem?

it is not that hard to rename channels in a raw file so that they match
names in a given montage.

I think we'd rather let people rename the montage names and apply than
change channel names then apply. You shouldn't need to change your Raw
instance channel names (which could be meaningful or standard for your
system) to get positions added. But this is perhaps a different issue.

cbrnr, 2017-08-19:
You're right, but maybe setting `transform=True` is not such a good idea if fiducials are required.

> I somehow missed this and it's not easy to check ATM (on mobile), do each
of the two listed in the top comment have this problem?

Yes, each of the two montages have this problem.

larsoner, 2017-08-21:
> maybe setting transform=True is not such a good idea if fiducials are required.

My proposal is to add fiducials to any montage that doesn't currently have them, and do `transform=True` as default via deprecation. In the long run this will facilitate a lot of things, such as template-based EEG localization.

Regarding the duplicate channel names, I guess we could split them up or deduplicate somehow. But I wonder if it's actually doing any damage or creating problems beyond not being the most elegant solution. If it's not causing problems, my vote would be just to document the redundancies.

cs50-c123b, 2017-08-21:
> My proposal is to add fiducials to any montage that doesn't currently have them, and do `transform=True` as default via deprecation. In the long run this will facilitate a lot of things, such as template-based EEG localization.

+1 if there is an informative error message in case the transform fails (such as "Required fiducials LPA, RPA, and Nz not found in montage. You can still load the montage by setting transform=False.")

> Regarding the duplicate channel names, I guess we could split them up or deduplicate somehow. But I wonder if it's actually doing any damage or creating problems beyond not being the most elegant solution. If it's not causing problems, my vote would be just to document the redundancies.

It's just ugly/unprofessional if you visualize them. If you happen to use these (non-standard) location names in your recordings, you should be prepared to use a custom montage. Or let's split it up into `standard_postfixed_a` and `standard_postfixed_b`.

cbrnr, 2017-08-21:
> My proposal is to add fiducials to any montage that doesn't currently have them, and do `transform=True` as default via deprecation. In the long run this will facilitate a lot of things, such as template-based EEG localization.

+1 if there is an informative error message in case the transform fails (such as "Required fiducials LPA, RPA, and Nz not found in montage. You can still load the montage by setting transform=False.")

> Regarding the duplicate channel names, I guess we could split them up or deduplicate somehow. But I wonder if it's actually doing any damage or creating problems beyond not being the most elegant solution. If it's not causing problems, my vote would be just to document the redundancies.

It's just ugly/unprofessional if you visualize them. If you happen to use these (non-standard) location names in your recordings, you should be prepared to use a custom montage. Or let's split it up into `standard_postfixed_a` and `standard_postfixed_b`.

teonbrooks, 2017-08-21:
> @teonbrooks any idea why we default to transform=False in our montage-reading code? Everything else in MNE is in Neuromag head coords, seems like we'd want to (at least by default) make them in head coords, too.
I agree, I think not all of the montages had fiducials so we left it there by default. also I think we just left it because we assumed the neuromag transformation was only needed when going to source space and that the user might want the electrode displayed in their native space. 

@larsoner, separate issue, but I think we have finally gotten to the point where we need to scope out a plan for restructuring the montage/digmontage class. also, make a straightforward procedure to go from a standard template to an average source brain

larsoner, 2017-08-22:
> +1 if there is an informative error message in case the transform fails

If we do the point-adding I'm advocating, then the transform really shouldn't ever fail, at least for our builtin montages (the way `read_montage` is really meant for public consumption as far as I can tell). But we can add it just to be safe. I think the check is there, we'll just need to tweak the message.

> It's just ugly/unprofessional if you visualize them.

Ahh, I hadn't thought about `montage.plot`. This would happen there. However, renaming is a rough API break.

Regardless of what we do with the montages, we should probably do something intelligent with numerically-identical location plotting (e.g., separate by newlines). Even for MEG sensor plotting like Neuromag, plotting their actual physical locations you'd want to do something like this. And maybe it would be good enough to make the `montage.plot` look okay with these FieldTrip montage files.

> @larsoner, separate issue, but I think we have finally gotten to the point where we need to scope out a plan for restructuring the montage/digmontage class.

There has been [a lot of discussion on this](https://github.com/mne-tools/mne-python/issues/3893#issuecomment-271925975) and I think people are mostly happy with the API, [you included based on a quick look](https://github.com/mne-tools/mne-python/issues/3893#issuecomment-272071754). Feel free to look it over again though.

> also, make a straightforward procedure to go from a standard template to an average source brain

Yep this is part of the motivation for #3893. That plus `mne.coreg` should get you there, and we can hopefully build a tutorial out of it.

cbrnr, 2017-08-22:
> If we do the point-adding I'm advocating, then the transform really shouldn't ever fail, at least for our builtin montages (the way `read_montage` is really meant for public consumption as far as I [...]

I think `read_montage` is also meant to read custom montage files, and without a clear error message it will be difficult to figure out what's going on when reading a montage without the required fiducials.

> Regardless of what we do with the montages, we should probably do something intelligent with numerically-identical location plotting (e.g., separate by newlines).

You mean `montage.plot` should check for identical locations and only plot one? Which label should it use then? The first? I guess this is feasible, and it could also throw a warning if it tries to visualize a montage with duplicate locations.

larsoner, 2017-08-22:
>  Which label should it use then? The first?

I was thinking of showing all names, separated by newlines. We'd have to do some testing to figure out the best option.

cbrnr, 2017-08-22:
Ah OK, might get a bit crowded then. What if there are more than 2 names, like 3 or 4? I guess we could also just leave it as it is. The plot then indicates that there's something wrong with the montage (which I still think is the case, at least for EEG there shouldn't be montages with duplicate names).

cbrnr, 2017-09-20:
I guess setting `transform=True` by default can be done in a new PR.

Besides that, what do we do with the two montages that have multiple labels for identical locations? People will probably start plotting montages because there now is a method for that and notice that something is wrong. I'd either remove these two montages or de-duplicate the labels.

larsoner, 2017-09-20:
I am against those for the same reasons as before (maintainability and
backward compatibility), and still in favor of the less invasive change of
making the plotting function smarter to handle overlaps.

cbrnr, 2017-09-20:
Then I propose to issue a warning if plotting such a montage and only plotting the first label. Handling more than one label per position will be difficult since there is already overlap with only one label per location.

larsoner, 2017-09-20:
That sounds reasonable to me.

cbrnr, 2017-09-21:
Checking for duplicate labels is giving me a headache. First, I need to use `scipy.spatial.distance.pdist` and `scipy.spatial.distance.squareform` - MNE does not require SciPy, so how should this be handled? Second, figuring out if there are duplicate labels is easy, but finding their indices and plotting only one label is pretty hard. I'm really not sure if these two defect montages are really worth this effort. In the end, they are defective and I still think we should either remove them completely or remove the duplicate labels from the montages.

larsoner, 2017-09-21:
> First, I need to use scipy.spatial.distance.pdist and scipy.spatial.distance.squareform

We have a standard fast pairwise distance wrapper `_compute_nearest(..., return_dists=True)`.

> MNE does not require SciPy, so how should this be handled?

SciPy is a hard dependency of MNE

> Second, figuring out if there are duplicate labels is easy, but finding their indices and plotting only one label is pretty hard.

Does this work?
```
use_pos = []
use_labels = []
usable = np.ones(len(pos), bool)
dists = _compute_nearest(pos, pos, return_dists=True)[1]
for ii in range(len(pos)):
    if usable[ii]:
        use_pos.append(pos[ii])
        use_labels.append(labels[ii])
    usable[dists[ii] < 1e-5] = False
logger.info('Plotting %d/%d deduplicated labels' % (len(use_pos), len(pos))
assert not usable.any()
```

cbrnr, 2017-09-21:
Thanks @larsoner. If SciPy is a hard dependency then I just use it. MNE's `_compute_nearest` returns a vector of all zeros, whereas `scipy.spatial.distance.cdist` works. I think I just found a solution without loops.

larsoner, 2017-09-21:
> _compute_nearest returns a vector of all zeros

Oh right because it gives the neighbor as zero. `cdist` should be fine

larsoner, 2017-09-21:
One non-(explicit-)loop would be to turn the square distance matrix into a connectivity matrix and do:
```
_, components = sparse.csgraph.connected_components(cdist(pos, pos) < 0.05)
```
And then unpack the first element of each item in `components`. Is that what you're doing?

cbrnr, 2017-09-21:
Here's what I'm doing right now:

```
dists = cdist(montage.pos, montage.pos)
dists[np.tril_indices(dists.shape[0])] = np.nan
dupes = np.argwhere(np.isclose(dists, 0))
```
This gives me all duplicates in the rows, column 0 is the index of the first label and column 1 the index of the second (duplicate) label. Now I'm working on integrating all of this into the plot function (which is not that straightforward because `Montage` objects don't support removal of channels out of the box, but anyway I'll do it manually).

cbrnr, 2017-09-21:
Let's see if this works in #4592...

PR #4592 (2017-09-25): [MRG] Do not plot duplicate labels in montage
Fix #4403.
closed 2017-09-25 by larsoner