mne-tools/mne-python #4407: Issues with CSP example

jasmainak, 2017-07-19:
I noticed a number of issues here with the [CSP example](https://martinos.org/mne/stable/auto_examples/decoding/plot_decoding_csp_eeg.html#sphx-glr-auto-examples-decoding-plot-decoding-csp-eeg-py)

* It's using deprecated parameters: 'firwin2' and 'norm_trace=True'. In fact, the warning for the 'norm_trace' left me confused for 5 minutes because it was talking about a parameter that didn't actually exist. Finally, I figured it must be the one which is boolean.

* If I use the recommended parameters, the patterns now look quite different. 

[image]

as opposed to

[image] on the website

Note the scale and see the last pattern.

* Finally, I was a bit mystified that 5 patterns were being plotted even though 4 components were estimated. The documentation tells me that the shape of patterns_ should be n_components x n_channels but I see that the shape is n_channels x n_channels

Sorry I don't have time to look into more detail at the moment but hopefully someone can take over

jasmainak, 2017-07-19:
Also, I suspect the TimeDecoding class could be used to simplify the example quite a bit

jaeilepp, 2017-07-20:
Most of the issues are already fixed in [dev](http://martinos.org/mne/dev/auto_examples/decoding/plot_decoding_csp_space.html?highlight=csp).
The number of components seems weird. Looks like only the number of filters is limited in the code. I suppose the ``patterns_`` should be as well.

jasmainak, 2017-07-20:
See [here] (https://github.com/mne-tools/mne-python/blob/a2b75f7d06fd753b7732f71a1f92802abc28dd63/mne/decoding/csp.py#L131). The warning is also still broken and I see `norm_trace` is not set to True also in the dev version.

closed 2017-08-02 by Eric89GXL