mne-tools/mne-python #4410: bad SPM trans file

agramfort, 2017-07-20:
we've done some digging with @britta-wstnr  and it appears that the spm trans file we ship is bad:

[image]

with this new trans file 

[spm-ok-trans.fif.zip](https://github.com/mne-tools/mne-python/files/1163567/spm-ok-trans.fif.zip)

I get:

[image]

which looks much better. Also it seems like that spm ships the dig points with the datasets and we don't. That could guarantee that our coreg is great.

@jaeilepp I think you did this initially. do you remember anything about this?

jaeilepp, 2017-07-21:
No, I don't think I've touched the spm dataset.

agramfort, 2017-07-21:
indeed. It seems it's denis, eric and me who wrote most of it :)

https://github.com/mne-tools/mne-python/blame/master/examples/datasets/plot_spm_faces_dataset.py

Eric89GXL, 2017-07-21:
> it appears that the spm trans file we ship is bad

I think you owe an apology to the `plot_trans` function, who was inappropriately blamed for this bad alignment image last week. :)

So you used `mne.coreg` to do it and everything looks good now? Then we should update the dataset. Would be a good time to update the `sample` dataset for #4383. @dengemann can you help?

agramfort, 2017-07-21:
>
> I think you owe an apology to the plot_trans function, who was
> inappropriately blamed for this bad alignment image last week. :)
>

:)

> So you used mne.coreg to do it and everything looks good now?
>
yes except I don't have the dig points to align precisely.

> Then we should update the dataset. Would be a good time to update the
> sample dataset for #4383
> <https://github.com/mne-tools/mne-python/pull/4383>. @dengemann
> <https://github.com/dengemann> can you help?
>

what needs to do on sample?

Eric89GXL, 2017-07-21:
`left_right` maps for `fsaverage` from FreeSurfer 6

agramfort, 2017-07-21:
won't have time to do this before 10 days. we need to add the dig points to
SPM package

agramfort, 2017-07-24:
we should also add to the sample datasets a good annotation file with blinks, flux jumps etc. There are a few...

jaeilepp, 2017-07-28:
Can I get a list of things that need to be updated? I can take a look next week.

agramfort, 2017-07-28:
add the dig points + a fixed trans file

nothing else pops to mind

jaeilepp, 2017-07-31:
The dataset [here](http://www.fil.ion.ucl.ac.uk/spm/data/mmfaces/) has the ``head_shape.mat`` file from which I was able to extract the fiducials and coregistrate with. Here's the result:
[image]

I could upload the raw data with those fiducials (605 extra points). I suppose there is not a standard for those files, so the raw reader stays as it is.

agramfort, 2017-07-31:
looks great

> I could upload the raw data with those fiducials (605 extra points).

what do you mean by upload? already on Amazon? in a separate fif file? in
the .ds files?

jaeilepp, 2017-07-31:
I was thinking that I'll upload a separate fif-file and also the ``head_shape.mat`` to the amazon. I had to do some tweaks to append the digs, so I guess this cannot really be generalized to the ``read_raw_ctf``.

agramfort, 2017-07-31:
+1 for a tiny fif files that contains dig points + the headshape.mat file.

how does the code on the spm tutorial look like? it's not too much code
to be pasted in examples?

jaeilepp, 2017-07-31:
You mean to add a call of ``plot_trans`` with the fiducials?

agramfort, 2017-07-31:
I mean do you end up messing with custom files and transformations in our examples ? If not lets push the new dataset on amazon and update examples

closed 2017-07-31 by agramfort