mne-tools/mne-python #4412: switch travis + doc from nosetest to pytest

agramfort, 2017-07-21:
@massich that can also be an easy thing to get started.

massich, 2017-07-26:
For compleatness, some link references:
* [scikit-learn migration issue](https://github.com/scikit-learn/scikit-learn/issues/7319)
* [scikit-learn first PR](https://github.com/scikit-learn/scikit-learn/pull/8246)
* [nose converter](https://github.com/pytest-dev/nose2pytest)
* [unit test converter](https://github.com/pytest-dev/unittest2pytest)
* [unit test converter](https://github.com/dropbox/dbx-unittest2pytest) (another)
* [pytest examples](https://docs.pytest.org/en/latest/example/simple.html)

Eric89GXL, 2017-07-26:
SciPy also just moved over in the last couple of weeks

massich, 2017-07-26:
@Eric89GXL good to know. I'll check how they did it. (unless you have already gone through their **Issues** and **PRs**, and want to list them yourself here)

Eric89GXL, 2017-07-26:
The SciPy PRs were by user "pv"

closed 2017-08-16 by larsoner