mne-tools/mne-python #4415: epochs.info throw an error

jsbyysheng, 2017-07-23:
I used code in [http://mne-tools.github.io/stable/auto_examples/io/plot_objects_from_arrays.html#sphx-glr-auto-examples-io-plot-objects-from-arrays-py](url) and wanted to see the variable  `epochs.info` but I got an err like: 
`File "C:\ProgramData\Anaconda3\lib\site-packages\mne\io\meas_info.py", line 204, in __repr__
    entr = dt.fromtimestamp(v[0]).strftime('%Y-%m-%d %H:%M:%S')
OSError: [Errno 22] Invalid argument`
My python version is 3.6 and mne is 0.14.1

agramfort, 2017-07-23:
Please share a gist to replicate

Alex

jsbyysheng, 2017-07-23:
for example:
```
import numpy as np
import neo
import mne

sfreq = 1000  # Sampling frequency
times = np.arange(0, 10, 0.001)  # Use 10000 samples (10s)
sin = np.sin(times * 10)  # Multiplied by 10 for shorter cycles
cos = np.cos(times * 10)
sinX2 = sin * 2
cosX2 = cos * 2
# Numpy array of size 4 X 10000.
data = np.array([sin, cos, sinX2, cosX2])
# Definition of channel types and names.
ch_types = ['mag', 'mag', 'grad', 'grad']
ch_names = ['sin', 'cos', 'sinX2', 'cosX2']
info = mne.create_info(ch_names=ch_names, sfreq=sfreq, ch_types=ch_types)
raw = mne.io.RawArray(data, info)
print(raw.info)
```
These codes works well in ipython on my ubuntu server but throw an error in my windows10 laptop. I use anaconda in my laptop and edit with spyder.

agramfort, 2017-07-23:
can you set the LC_CTYPE env variable to 'en_US.UTF-8'

import os
os.environ['LC_TYPE'] = 'en_US.UTF-8'

and tell me this fixes the pb?

jsbyysheng, 2017-07-24:
have no effect. I also tried the code above using python.exe directly and it has no effect also.

agramfort, 2017-07-24:
can you print os.environ on your system?

jsbyysheng, 2017-07-24:
It's too long to show it here. I wonder if you want to confim this?
```
'C:\\ProgramData\\Anaconda3\\lib\\site-packages', 'TERM': 'xterm-color', 'CLICOLOR': '1', 'PAGER': 'cat', 'GIT_PAGER': 'cat', 'MPLBACKEND': 'module://ipykernel.pylab.backend_inline', 'LC_TYPE': 'en_US.UTF-8'})
```
`'LC_TYPE': 'en_US.UTF-8'`

agramfort, 2017-07-24:
I really suspect a non-english US env var that breaks datetime. but I can
be wrong

can you make a minimal snippet to replicate. Somethign reusing:

dt.fromtimestamp(v[0]).strftime('%Y-%m-%d %H:%M:%S')

jsbyysheng, 2017-07-24:
I think I've found the reason.
```
from datetime import datetime as dt
#1970-01-02 08:00:00
print(dt.fromtimestamp(86400).strftime('%Y-%m-%d %H:%M:%S'))
#1970-01-02 07:59:59
print(dt.fromtimestamp(86399).strftime('%Y-%m-%d %H:%M:%S'))
```
Time which is less than `1970-01-02 07:59:59` is invalid in Windows10 but valid in linux. And the default meas_date in mne 0.14.1 is 1970-01-01 08:00:00. So if u try to `print(raw.info)` or `print(epoch.info)`， `OSError: [Errno 22] Invalid argument` can be threw.

agramfort, 2017-07-24:
let's make the default time : 1970-01-02 08:00:00 ok?

can you try and if it fixes the pb, send us a PR on the current dev version?

massich, 2017-07-26:
I runned into the same issue, while playing with nosetests and pytests.

larsoner, 2018-04-04:
I think this should be fixed on `master` (can't remember the PR offhand) but feel free to reopen if not

closed 2018-04-04 by larsoner