mne-tools/mne-python #4434: Anti-aliasing in plots

cbrnr, 2017-07-28:
I've noticed that most plots in MNE use the MPL default of `antialiasing=True` except for `plot_raw`, which specifically disables antialiasing. I'd argue that anti-aliasing makes for a much nicer look (for example `plot_pds` is a really nice representation of the spectrum for all channels), so I'd enable it also in `plot_raw`. What is the reason that this was disabled? With the current behavior, lines appear rather thick on some platforms, which doesn't appear to be the case if antialiasing is turned on.

jaeilepp, 2017-07-28:
It's purely for performance. It's still a bit slow with data that has many channels. That's why decim parameter was introduced.

cbrnr, 2017-07-28:
Hmmm. You probably know what I'm going to ask now :wink:

Any chance we could make this configurable (doesn't have to be public)?

jaeilepp, 2017-07-28:
Fine by me. Just occurred to me that this (and the escape button thing) could also be done in the mne config file.

cbrnr, 2017-07-28:
Yes, why not. Could you give me a quick pointer where I need to look for this?

jaeilepp, 2017-07-28:
You should have a ``.mne`` folder in your home directory.

cbrnr, 2017-07-28:
Alright. What's the preferred way to do this? I just put `mne.get_config('close_key', 'escape')` everywhere I need this value? Do we ever write it to the config file? If not I don't know if I really like this approach better, because this would mean that to change the default behavior I'd explicitly need to change the config file externally.

Also, currently the only thing stored in the config file is the window size for `plot_raw` in my case.

agramfort, 2017-07-28:
can we do it as for escape_key ie wihtout adding a new param?

cbrnr, 2017-07-28:
Oh yes, definitely - this would be my preferred option.

cbrnr, 2017-07-28:
Also, I just checked if there is any performance difference on my machine between antialiasing on and off. It's not noticeable, but `antialiasing=True` (combined with a line width of around `0.5`) looks much nicer.

cbrnr, 2017-07-28:
Anti-aliasing off, line width 0.1:
[image]

Anti-aliasing on, line width 0.5:
[image]

If there's no clear evidence that anti-aliasing negatively impacts performance, we should discuss using anti-aliasing in `plot_raw` by default like in all other plots.

Eric89GXL, 2017-07-28:
I can confirm on OSX at least that with `Qt4Agg` backend it makes little difference (~5% speed loss) when toggling `antialiased=True` in `mne/viz/raw.py` and using:
```
import time
import matplotlib
matplotlib.use('QtAgg')
import matplotlib.pyplot as plt
import mne
plt.ion()
raw = mne.io.read_raw_fif(mne.datasets.sample.data_path() + '/MEG/sample/sample_audvis_raw.fif', preload=True)
fig = raw.plot()
t0 = time.time()
for ii in range(20):
    fig.canvas.key_press_event('right')
print(time.time() - t0)
plt.close(fig)
```
It would be good if someone could also confirm on Linux (no access to that machine for me currently).

cbrnr, 2017-07-31:
Here are some results from running the test on Arch Linux and Windows 10 (in a VM with Arch Linux as host). The column shows the average time (in s) over 5 repetitions.
```
macos MacOSX antialiased=False       0.791953
macos MacOSX antialiased=True        0.797703
macos Qt5Agg antialiased=False       5.640340
macos Qt5Agg antialiased=True        5.796852
Arch Qt5Agg antialiased=False       15.258201
Arch Qt5Agg antialiased=True        15.278452
Windows Qt5Agg antialiased=False     5.278124
Windows Qt5Agg antialiased=True      5.259347
```

Apparently, anti-aliasing does not have a negative performance impact.

Besides, the numbers are interesting for other reasons as well. For example, the MacOSX backend is about 7x faster than Qt5Agg. On Windows, the Qt5Agg backend is about as fast as on macOS, but on Linux it is about 3x as slow (ironically Windows runs in a VM on Linux). @Eric89GXL do you think your test measures performance of backends, or are there other factors that influence the results? I'm wondering if this should be reported to the MPL devs...

agramfort, 2017-07-31:
should we recommend certain backends then?

+1 for antialiased=True by default

cbrnr, 2017-07-31:
> should we recommend certain backends then?

I'm not sure if these numbers allow comparing backends. I don't notice any difference between `Qt5Agg` and `MacOSX` in interactive plotting despite the 7x difference. If anything, I always felt that `Qt5Agg` was faster in some applications, such as plotting (or rather rotating) a 3D montage plot.

> +1 for antialiased=True by default

Just plotted the example data, this looks so much better. Let's make really sure that despite @Eric89GXL's and my results we do not impact performance (significantly) though.

Eric89GXL, 2017-07-31:
It probably measures mostly backend performance, yeah. Also note I excluded
the first draw, not sure if the initial draw changes things.

+1 for reenabling AA, thanks for checking OSes

cbrnr, 2017-07-31:
OK, so I guess I'll create a short PR that re-enables AA.

But if these numbers are true, I wonder why the MacOSX backend is so much faster than Qt5Agg (quite the opposite of my experience). Furthermore, why is the performance of Qt5Agg so bad on Linux (granted it can't be compared directly to the performance on a Mac because these are 2 different machines, but the Windows test ran on the same machine and is 3x faster). Should we worry/panic :smile: (i.e. drop a message to the MPL devs) or just ignore it?

Eric89GXL, 2017-07-31:
Let's check on a non-VM Linux before thinking too hard about it. Could be
an OSMesa problem or so

cbrnr, 2017-07-31:
Linux is the host and Windows is the guest!

Eric89GXL, 2017-07-31:
Well that's not good. It probably is worth investigating. I'd probably start by eliminating elements from our plot to see if any of them make it faster. (I'd actually start with not plotting any data lines, and see if that makes it take almost no time -- that tells us if it really is the data plotting that is bad.) Then we'd want to make some minimal example that doesn't rely on MNE. And at some point we'd need to test the latest version of `matplotlib`. Depending on how easy that is, that could even be the first step.

I could run some of these tests on my Linux machine next week.

cbrnr, 2017-07-31:
At least it's not related to anti-aliasing, so I think it's safe to merge #4442. Also, I don't _notice_ any performance difference, so I think the test procedure might be flawed. I think I have the latest MPL release, but we should test with master. I'll ask them to see if it's worth investigating at all. I think it's best if I open a new issue and close this one.

closed 2017-07-31 by cbrnr

Eric89GXL, 2017-08-08:
For the example code I posted above, on my Linux system Qt4Agg / Qt5Agg / WXAgg all take about 2.8 seconds, and TkAgg takes about 5. I'm on `matplotlib` 2.02. But in any case, perhaps your Arch linux probably has some optimization issues.

Eric89GXL, 2017-08-08:
(I'm on proprietary NVIDIA drivers over here, maybe you're using a slower graphics setup?)

PR #4442 (2017-08-01): [MRG] Enable anti-aliasing in plot_raw
Fix #4434