mne-tools/mne-python #4438: problem computing nave in contrast

jasmainak, 2017-07-29:
@jaeilepp my parting gift for you ;)

Here is the script to reproduce

```py
import mne
from mne.datasets import sample
from mne import read_evokeds

path = sample.data_path()
fname = path + '/MEG/sample/sample_audvis-ave.fif'

condition = 'Left Auditory'
evoked = read_evokeds(fname, condition=condition, baseline=(None, 0))
contrast = mne.combine_evoked(
    [evoked, evoked, evoked], weights=[0.5, 0.5, -1.])
print('Nave in contrast: %d' % contrast.nave)
print('Total number of epochs: %d' % (evoked.nave * 3))
```
Note that I didn't find any documentation for what `nave` should actually be, but I assumed it should be the addition of number of epochs in each evoked?

cc @dengemann @agramfort

agramfort, 2017-07-29:
code in master is actually correct. After a bit of thoughts you get a
reduction of nave because the weights vector is not of norm 1. See:

```
import numpy as np
from scipy import linalg

import mne
from mne.datasets import sample
from mne import read_evokeds

path = sample.data_path()
fname = path + '/MEG/sample/sample_audvis-ave.fif'

condition = 'Left Auditory'
evoked = read_evokeds(fname, condition=condition, baseline=(None, 0))

contrast = mne.combine_evoked(
    [evoked, evoked], weights=[0.5, 0.5])
assert contrast.nave == evoked.nave * 2

contrast = mne.combine_evoked(
    [evoked, evoked], weights=[0.5, -0.5])
assert contrast.nave == evoked.nave * 2

weights = np.array([0.5, 0.5, -1.])
weights /= linalg.norm(weights)

contrast = mne.combine_evoked(
    [evoked, evoked, evoked], weights=weights)

assert contrast.nave == evoked.nave
```

jasmainak, 2017-07-29:
So, this should be [0.25, 0.25, -0.5] you mean? Why not throw an error if it's not the case?

jasmainak, 2017-07-29:
> assert contrast.nave == evoked.nave

why this and not 3 * evoked.nave? What does the `nave` mean?

agramfort, 2017-07-29:
can you make sense of
https://github.com/mne-tools/mne-python/blob/master/mne/evoked.py#L920 ?

for me there is no bug in combine_evoked

try playing with np.random.randn

if you do:

np.std(a * np.random.randn(1000) + b * np.random.randn(1000))

changing a and b what do you get?

jasmainak, 2017-07-29:
I need to sleep over this. If I do:

```py
evoked_scaled = mne.combine_evoked([evoked], [0.5])
print(evoked.nave)
print(evoked_scaled.nave)
```
it gives me 55 and 220. But now if I do:

```py
evoked_scaled = evoked.copy()
evoked_scaled.data *= 0.5
print(evoked.nave)
print(evoked_scaled.nave)
```
I get 55 and 55. Why should scaling change `nave`?

larsoner, 2017-08-15:
`nave` reflects how much the Gaussian noise level (computed e.g. using a covariance) will be reduced by averaging trials together.

If you scale down your data by some factor (here, 0.5), then the noise level goes down by that factor (here, 0.5), which is the same noise level that would have resulted from averaging `1/0.5 ** 2 == 4` times as many trials together, which is why `nave` quadruples.

closed 2017-08-15 by larsoner

jasmainak, 2017-08-15:
yes but if I understand correctly that this affects rank estimation in `plot_white` ...

dengemann, 2017-08-15:
Don't worry, there is a different type of fix scheduled. Now that we
cracked the big autoreject nut I may soon find time to push that.

jasmainak, 2017-08-15:
okay great :)