mne-tools/mne-python #4446: mne. grade_to_tris function for VolumeSourceSpace

d-van-de-velden, 2017-08-02:
Dear Sir or Madame,

I am performing Volume Source Localization and I plan on using a spatiotemporal clustering test on it with the mne package.

Therefore I need to define the neighbor vertices for each vertice. In 'ordinary' source space one use the mne.grade_to_tris function. But it isn't able to be used for a volume source space. 
Is there a alternative or is the solution to the problem up to myself ?

Yours sincerely,
Daniel van de Velden

agramfort, 2017-08-02:
this is not implemented. I fear you'll have too look into it yourself.

d-van-de-velden, 2017-08-03:
I found the 'mne.spatial_src_connectivity' function. This function is also not capable of performing a neighbor connectivity calculation, isn't it ?

Would you mind giving me a little help with the solution ? Maybe you are also interested in the solution.

joewalter, 2017-08-03:
Since you know the spacing of your volume source space, isn't it sufficient to simply compute the euclidean distance from one source to all other sources and take these as neighbours which are closer than (spacing * 1.1)

agramfort, 2017-08-04:
maybe this can help:

https://gist.github.com/agramfort/a3e3139a1f64315885c923a57e541e7a

please let us know if it does so we can add an example to the mne website.

d-van-de-velden, 2017-08-04:
Thanks a lot for the hint/help.
Right now I am processing and investigating the subject data. The solution of the mentioned problem will come up later. 

Of course I will let you know about the solution. Posting the solution in this thread as an attached file is fine ?

agramfort, 2017-08-04:
yes it will be good to start with

d-van-de-velden, 2017-08-22:
Dear all,

since my computation and Volume Source Estimation is done, I am moving forward to statistical testing - Spatio temporal Clustering Test. 
My group study also uses the "fsaverage" brain to make a reasonable comparison of the results. Therefore, its necessary to morph my mne.VolumeSourceEstimate from one subject to an other. Is there any function I wasn't able to find? If there is no function, has anyone ever tried to  do it or tried to 'hack' the surfaces source morphing to accept volume source estimate?

Since it is a delicate matter and I am not an expert yet, I appreciate any corporation.
Greetings,
Daniel

P.S.: I think, since this problem falls in the process of this threads solution, this post is correctly placed.

agramfort, 2017-08-22:
morphing volume data is not supported and I personally never did it.

you'd better ask on the MNE mailing list for this.

I suspect you'll need FSL or AFNI or ANTS or SPM softwares for this.

larsoner, 2018-04-04:
We now have a function for volume neighbors / connectivity, so closing this one

closed 2018-04-04 by larsoner

d-van-de-velden, 2018-04-04:
Hello,

that is good news. May I ask for a reference to said function ?
Moreover, how are you verified the correctness of the function? I am asking, since I am going to use it in my project and its not used often enough to be sure about it. 

Thanks in advance,
Daniel

larsoner, 2018-04-04:
See https://github.com/mne-tools/mne-python/pull/4544

d-van-de-velden, 2018-04-04:
So to make it clear. You are referring to "def _spatio_temporal_src_connectivity_vol(src, n_times)" ?

larsoner, 2018-04-04:
No you should be able to use the public function `spatio_temporal_src_connectivity`