mne-tools/mne-python #4450: Combining / Extracting Epochs

seantrott, 2017-08-02:
Hello,

I am working with EEGLAB (".set") data, and I'd like to do some analyses with MNE. I am able to read in the data, using this helpful tutorial (http://martinos.org/mne/dev/manual/migrating.html), and I'd like to aggregate the data to do a time decoding analysis (as discussed here: http://martinos.org/mne/stable/auto_tutorials/plot_sensors_decoding.html). 

Unfortunately, each event is epoched with a separate event code, instead of each being either '1' or '2'. Basically, I need to:
1. Get all the events of interest.
2. Aggregate/label each event / epoch as the correct condition, e.g. '1' or '2'.

Currently, I see two ways I could do this:

One, I could use "combine_event_ids" on all of the possible epochs.

`epochs = mne.Epochs(raw, events)  
epochs = mne.epochs.combine_event_ids(epochs, s, 11, copy=True)  
epochs = mne.epochs.combine_event_ids(epochs, l, 22, copy=True)`

(Where 's' and 'l' the list of event_ids I want to combine into the new id). For this approach, is there a way to get rid of all the previous epochs, e.g. the ones labeled with separate event codes? Additionally, can I label the new ones with something other than integers?

The other way would be on the initialization of the "Epochs", filtering which event ids are labeled.  Ideally, I could do something like this:

`event_id2 = dict(cond1=list1, cond2=list2)  
epochs = mne.Epochs(raw, events, event_id)
`
But I've looked into the code, and it seems like this isn't an option. I could do some tinkering with the code, but I'm hesitant to change much.

Which way would be the best approach?

Please let me know if you need more information. Thank you!
-Sean Trott

jaeilepp, 2017-08-03:
The first approach looks good to me. You can use dicts as the event id. So basically you can do 
```
epochs = mne.epochs.combine_event_ids(epochs, s, {'First': 11}, copy=True)
epochs = mne.epochs.combine_event_ids(epochs, l, {'Second':22}, copy=True)
```
Then it is also possible to combine them to a single object using ``mne.concatenate_epochs``. After that you can access them separately by doing for example ``epochs['First']``.
Hope this helps.

seantrott, 2017-08-03:
Thanks, it looks like this worked!

closed 2017-08-03 by seantrott