mne-tools/mne-python #4452: Failing test in test_source_space.py

joewalter, 2017-08-03:
`nosetests mne/tests/test_source_space.py:test_other_volume_source_spaces` fails for me locally with the current master. Can someone confirm?

jaeilepp, 2017-08-03:
It passes for me. Do you have MNE-C installed?

agramfort, 2017-08-03:
it does not pass for me either:

F
======================================================================
FAIL: Test setting up other volume source spaces.
----------------------------------------------------------------------
Traceback (most recent call last):
  File
"/Users/alex/anaconda/envs/py35/lib/python3.5/site-packages/nose/case.py",
line 198, in runTest
    self.test(*self.arg)
  File
"/Users/alex/anaconda/envs/py35/lib/python3.5/site-packages/numpy/testing/decorators.py",
line 147, in skipper_func
    return f(*args, **kwargs)
  File "/Users/alex/work/src/mne-python/mne/utils.py", line 843, in dec
    return function(*args, **kwargs)
  File "/Users/alex/work/src/mne-python/mne/tests/test_source_space.py",
line 257, in test_other_volume_source_spaces
    _compare_source_spaces(src, src_new, mode='approx')
  File "/Users/alex/work/src/mne-python/mne/source_space.py", line 2680, in
_compare_source_spaces
    assert_equal(s0[name], s1[name], name)
AssertionError: 8407 != 7497 : nuse
    '8407 != 7497' = '%s != %s' % _common_shorten_repr(8407, 7497)
    '8407 != 7497 : nuse' = self._formatMessage('8407 != 7497 : nuse',
'8407 != 7497')
>>  raise self.failureException('8407 != 7497 : nuse')

----------------------------------------------------------------------

do you have the same error?

joewalter, 2017-08-03:
@jaeilepp yes I have MNE-C
@agramfort I have the same error

jaeilepp, 2017-08-03:
It must be some recent change. I pulled the latest and it fails for me too.

jaeilepp, 2017-08-03:
Looks like it's https://github.com/mne-tools/mne-python/pull/4355

agramfort, 2017-08-03:
hum. How come it is not detected on travis?

Eric89GXL, 2017-08-03:
I'll try to look by early next week

Eric89GXL, 2017-08-04:
So it turns out this is failing because the updated code is actually *more* accurate at determining points inside the limit than it was before.

Why? The `_filter_source_spaces` code, which operates on tessalated surfaces, first filters out points outside the surface (using sum of solid angles). Of the remaining points, it finds the closest surface vertex, and ensures that it is at least `limit` mm away from this point. When the user requested a sphere, it used to use a sphere with an ico-3 surface, which has 642 points. This puts the points roughly 1cm apart on the 90mm radius sphere used in the test. Thus you can be up to ~5mm away from the nearest tesselation point *and actually be on the surface of the sphere*, yet such a point will be kept. In this test it's not that bad -- the point closest to the surface of the sphere is 3.5mm inside. This is clearly less than the 5mm it should be, but is not so terrible.

So maybe the new values should be used as a bugfix, and we should document it as an accuracy increase for sphere-based vertex limiting in volume source spaces?

Eric89GXL, 2017-08-04:
... as to why it wasn't tested, it looks like the `requires_mne` tests were skipped on that run, not sure why. Maybe in the test-split migration I accidentally got rid of those tests? I'll look.

Eric89GXL, 2017-08-04:
Turns out I broke it back in December (!) when migrating to CI-helpers (#3831).

Eric89GXL, 2017-08-04:
@agramfort should we use the (current) more accurate calculation and call it a bugfix?

agramfort, 2017-08-04:
@Eric89GXL can you explain what happened? why it broke?

Eric89GXL, 2017-08-04:
The explanation is [five posts up](https://github.com/mne-tools/mne-python/issues/4452#issuecomment-320143741)

agramfort, 2017-08-04:
Ok for bug fix then ;)

PR #4458 (2017-08-04): FIX: Fix MNE-C
This should first expose the error in #4452. Will add a fix once we decide how best to fix it.

Closes #4452.
closed 2017-08-04 by Eric89GXL