mne-tools/mne-python #4454: Add window argument to psd_welch?

cbrnr, 2017-08-03:
I was wondering if a new `window` argument might be useful for the `mne.time_frequency.psd_welch` function, where the user can specify the window type (now the SciPy default is used, which corresponds to a Tukey 0.25 window). I usually use a Hamming or Hanning window, not because I know they work better but just out of habit. For reference, MATLAB uses a Hamming window by default for its [`pwelch` function](https://www.mathworks.com/help/signal/ref/pwelch.html).

Eric89GXL, 2017-08-03:
Can you try playing with it locally first to see if it makes much
difference? I'd rather not add it if the practical implications are limited
for our data, and instead document that it's different from MATLAB.

cbrnr, 2017-08-04:
Yes, it will make a difference, as will all other possible choices of window functions. Although it would be nice to have an argument to specify the window function (as different people will have different needs), a sane default might be sufficient for EEG/MEG application. At the very least, the used window should be documented in the `psd_welch` function, which is currently not the case. Now that I've looked over the code again, I'm not even sure that the Tukey window is used - I think the window is set to a Hann window [here](https://github.com/mne-tools/mne-python/blob/master/mne/time_frequency/psd.py#L17), which overrides the SciPy `spectrogram` default. Correct?

What do other think? Is there a need to specify [window functions](https://en.wikipedia.org/wiki/Window_function)? If Hann is really used, this might be a sensible choice for most use cases.

agramfort, 2017-08-04:
I would favor a good default choice for EEG/MEG/iEEG data. If there is no
such things then we need to motivate / document it.

cbrnr, 2017-08-04:
Not sure if there any studies specific to electrophysiological PSD windowing functions. I thing that this is likely a very minor point, so let's at least document which window is being used in `psd_welch`. Can you confirm that it is indeed a Hann window?

Eric89GXL, 2017-08-04:
> Can you confirm that it is indeed a Hann window?

Yeah it looks like we use `hann`. If you want, we could switch it to `hamming` for consistency. I don't expect it will make much difference in the output. If you do want to change it, go ahead and do it and we can compare the CircleCI output for the sample dataset raw PSD to see if it's qualitatively different.

Documentation either way is appreciated

cbrnr, 2017-08-16:
OK, switched to Hamming - let's see how the tests work out (#4492).

PR #4492 (2017-08-16): MRG+2: Use Hamming window for psd_welch (and document it)
Fix #4454
closed 2017-08-16 by larsoner