mne-tools/mne-python #4457: mne.find_events seems to decrease each value of 1 !

brohaut, 2017-08-03:
Dear MNE users,
I found a strange behavior while trying to save my customized events in a raw.fif for further use :

I first injected some custom triggers coded as [1, 2, 3, 4] in STI 014, then I run mne.find_events and everything worked fine:

    trig_chan = mne.pick_channels(raw.info['ch_names'], include=['STI 014'])
    np.unique(raw._data[trig_chan[0],:])
    array([ 0.,  1.,  2.,  3.,  4.])

    events = mne.find_events(raw,stim_channel=['STI 014'])
    96 events found
    Events id: [1 2 3 4]

But then when I saved this raw object using : 

    raw.save('subject_x_raw.fif')
And replayed the same script after having loaded this .fif, I get this strange results:

    raw = mne.io.read_raw_fif('subject_x_raw.fif', preload=True)
    np.unique(raw._data[trig_chan[0],:])
    array([ 0.,  1.,  2.,  3.,  4.])

    events = mne.find_events(raw,stim_channel=['STI 014'])
    73 events found
    Events id: [1 2 3]

STI 014 still contained the good value (array([ 0.,  1.,  2.,  3.,  4.])) but mne.find_events seemed to decrease each value of 1 and I basically lose my "1" events...

I was wondering whether that could cause some issue... so I report it !

Sincerely
Benjamin

agramfort, 2017-08-04:
can you share a file + script to replicate this?

did you filter / decimate in some way the files before saving them back to
disk?

larsoner, 2019-08-20:
Closing, no response

closed 2019-08-20 by larsoner