mne-tools/mne-python #4467: [RFC] Undesired change in locale due to traitsui import.

massich, 2017-08-07:
Due to a difference between `pytest` and `nosetests` test collection I runned into the problem described in #4462. Which is nothing but this other problem #2859 (at that time the error messaging was improved, but the problem was not solved).

As a summary: there is a `setlocale` [since early versions of `Qt`](https://stackoverflow.com/questions/25661295/why-does-qcoreapplication-call-setlocalelc-all-by-default-on-unix-linux/26745357#26745357). This had been producing error such [this one in matplotlib](https://github.com/matplotlib/matplotlib/issues/1867). 

What we can do is: first, [create a context manager](https://stackoverflow.com/questions/18593661/how-do-i-strftime-a-date-object-in-a-different-locale) 
```py
import locale
import threading

from contextlib import contextmanager

LOCALE_LOCK = threading.Lock()

@contextmanager
def setlocale(name):
    with LOCALE_LOCK:
        saved = locale.setlocale(locale.LC_ALL)
        try:
            yield locale.setlocale(locale.LC_ALL, name)
        finally:
            locale.setlocale(locale.LC_ALL, saved)
```
and then, either [decorate](https://docs.python.org/2/library/functools.html#functools.wraps) the problematic tests ([like `test_find_ch_connectivity`](https://github.com/mne-tools/mne-python/blob/5a9fe0332a768334b0e200a6b5ce50f0af3c4805/mne/channels/tests/test_channels.py#L223)) to enforce the testing with the right locale + adding a test with a mismatched locale to cover [this line](https://github.com/mne-tools/mne-python/blob/d446fe79bc8bf6cadb7d8ab1209800b729b0c06a/mne/io/ctf/info.py#L82) 

```py
from functools import wraps

def my_decorator(f):
    @wraps(f)
    def wrapper(*args, **kwds):
        with setlocale('C'):
                return f(*args, **kwds)
    return wrapper

@my_decorator
```

Or, add the context manager right in the code base itself ([here](https://github.com/mne-tools/mne-python/blob/d446fe79bc8bf6cadb7d8ab1209800b729b0c06a/mne/io/ctf/info.py#L76)) rather than in the tests. (+ also covering the raise case).

I cannot decide which option is better. Any ideas @Eric89GXL @dengemann ?

dengemann, 2017-08-07:
Can we first try to understand what is happening here? It is reminiscent of the matplotlib issues we've had in the past. Nested imports have been the solution ever since. Maybe we can fix it with a less invasive and complex edit. As discussed, let's try to see if your dummy examples can be fixed by nesting imports.

massich, 2017-08-07:
I have checked. Actually in my dummy cases in #4462 nesting the import and force pytest to execute the tests in a different order, fixes it. Back to the mne code, the imports are actually nested already. However `mne/gui/__init__.py` has this function:

```py
def combine_kit_markers():
    """Create a new KIT marker file by interpolating two marker files."""
    _check_mayavi_version()
    from ._backend import _check_backend
    _check_backend()
    from ._marker_gui import CombineMarkersFrame
    frame = CombineMarkersFrame()
    return _initialize_gui(frame)
```
that calls `_marker_gui` which has the imports at file level.

dengemann, 2017-08-07:
@massich sounds interesting. But why does this import mayavi if the function is not called at the init level. I still don't understand it.

massich, 2017-08-07:
neither do I. yet.

jorisvandenbossche, 2017-08-07:
Because the pytest test discovery for some reason ran the gui/._marker_gui.py` file. The question is then why it did this (as following the documentation it should only search for `test_*.py` files). 
@massich did you try that with not running the doctests? as that is a reason to run each python file instead of just the test files.

massich, 2017-08-07:
Indeed, this is triggered by the doctests. And nosetests also have the same behavior.

massich, 2017-08-07:
just for completeness: 
```sh
LANG=en_US.UTF-8
LANGUAGE=en_US
LC_CTYPE="en_US.UTF-8"
LC_NUMERIC=fr_FR.UTF-8
LC_TIME=fr_FR.UTF-8
LC_COLLATE="en_US.UTF-8"
LC_MONETARY=fr_FR.UTF-8
LC_MESSAGES="en_US.UTF-8"
LC_PAPER=fr_FR.UTF-8
LC_NAME=fr_FR.UTF-8
LC_ADDRESS=fr_FR.UTF-8
LC_TELEPHONE=fr_FR.UTF-8
LC_MEASUREMENT=fr_FR.UTF-8
LC_IDENTIFICATION=fr_FR.UTF-8
LC_ALL=
```

Eric89GXL, 2017-08-07:
> either decorate the problematic tests (like test_find_ch_connectivity) to enforce the testing with the right locale + adding a test with a mismatched locale to cover this line

Decorating sounds like it will be annoying to add if we have to do it separately. Can we just build in into the `requires_mayavi` testing decorator (or if we only do `requires_module('mayavi')`, then make a mayavi-specific one)? Then it doesn't add any future additional work at the test-creation end.

massich, 2017-08-07:
> Can we just build in into the requires_mayavi testing decorator
It is not that the test requires mayavi. `test_find_ch_connectivity` does not require mayavi at all. The problem is that if you run something before that test (for instance the exploration of the doctests) that calls this line `from mayavi.core.ui.mayavi_scene import MayaviScene`, then the locale gets changed and the test fails.

Another option would to make mayavi 'safe' either by wrapping all mayavi functions and make sure that they don't change the environment or by doing a PR at mayavi.

larsoner, 2018-04-04:
Closing as an upstream bug.

Do you see this on our latest `environment.yml`?

closed 2018-04-04 by larsoner