mne-tools/mne-python #4469: morph_data error

romquentin, 2017-08-07:
Hi all, 
I am trying to morph individual source data to fsaverage. I am using this piece of code:
```
evoked = EvokedArray(patterns, epochs.info,
                                 tmin=epochs.tmin)
inv = read_inverse_operator(op.join(path_data, subject,
                                        subject + '-inv.fif'))
stc = apply_inverse(evoked, inv, lambda2=1.0 / (2 ** snr),
                                method='MNE', pick_ori='normal')
vertices_fsaverage = [np.arange(10242), np.arange(10242)]
morphed = mne.morph_data(subject, 'fsaverage', stc,
                                     n_jobs=1, grade=vertices_fsaverage,
                                     subjects_dir=subjects_dir)
```
It finished without error on only 2 participants out of 8. When it crashed, the error is this one.
```
Morphing data...
<ipython-input-32-bca94b1f110b>:33: RuntimeWarning: Morph map "/Users/quentinra/Desktop/MEGdata/WM1/subjects/morph-maps/fsaverage-sub12_XZUKLHYZ_AE-morph.fif" does not exist, creating it and saving it to disk (this may take a few minutes)
  subjects_dir=subjects_dir)
Morph map "/Users/quentinra/Desktop/MEGdata/WM1/subjects/morph-maps/fsaverage-sub12_XZUKLHYZ_AE-morph.fif" does not exist, creating it and saving it to disk (this may take a few minutes)
Creating morph map sub12_XZUKLHYZ_AE -> fsaverage
Creating morph map fsaverage -> sub12_XZUKLHYZ_AE
    21 smooth iterations done.
    21 smooth iterations done.
    21 smooth iterations done.
---------------------------------------------------------------------------
IndexError                                Traceback (most recent call last)
<ipython-input-32-bca94b1f110b> in <module>()
     31 morphed = mne.morph_data(subject, 'fsaverage', stc,
     32                          n_jobs=1, grade=vertices_fsaverage,
---> 33                          subjects_dir=subjects_dir)

/Applications/anaconda/lib/python2.7/site-packages/mne-python/mne/source_estimate.pyc in morph_data(subject_from, subject_to, stc_from, grade, smooth, subjects_dir, buffer_size, n_jobs, warn, verbose)

/Applications/anaconda/lib/python2.7/site-packages/mne-python/mne/utils.pyc in verbose(function, *args, **kwargs)
    724         with use_log_level(verbose_level):
    725             return function(*args, **kwargs)
--> 726     return function(*args, **kwargs)
    727 
    728 

/Applications/anaconda/lib/python2.7/site-packages/mne-python/mne/source_estimate.pyc in morph_data(subject_from, subject_to, stc_from, grade, smooth, subjects_dir, buffer_size, n_jobs, warn, verbose)
   2157                                      warn=warn)
   2158                      for data_buffer
-> 2159                      in np.array_split(data[hemi], n_chunks, axis=1)), axis=1)
   2160 
   2161     vertices = [nearest[0], nearest[1]]

/Applications/anaconda/lib/python2.7/site-packages/mne-python/mne/source_estimate.pyc in <genexpr>((data_buffer,))
   2156                                      n_vertices, nearest[hemi], maps[hemi],
   2157                                      warn=warn)
-> 2158                      for data_buffer
   2159                      in np.array_split(data[hemi], n_chunks, axis=1)), axis=1)
   2160 

/Applications/anaconda/lib/python2.7/site-packages/mne-python/mne/source_estimate.pyc in _morph_buffer(data, idx_use, e, smooth, n_vertices, nearest, maps, warn, verbose)

/Applications/anaconda/lib/python2.7/site-packages/mne-python/mne/utils.pyc in verbose(function, *args, **kwargs)
    724         with use_log_level(verbose_level):
    725             return function(*args, **kwargs)
--> 726     return function(*args, **kwargs)
    727 
    728 

/Applications/anaconda/lib/python2.7/site-packages/mne-python/mne/source_estimate.pyc in _morph_buffer(data, idx_use, e, smooth, n_vertices, nearest, maps, warn, verbose)
   1947         # get the row sum
   1948         mult = np.zeros(e.shape[1])
-> 1949         mult[idx_use] = 1
   1950         idx_use_data = idx_use
   1951         data_sum = e * mult

IndexError: index 167004 is out of bounds for axis 1 with size 166949
```
(index and size in the `IndexError` varies for each participant)
Would you have an idea on how to fix that?
Thanks a lot!
Romain

Eric89GXL, 2017-08-08:
Can you share these files somewhere?

* `/Users/quentinra/Desktop/MEGdata/WM1/subjects/morph-maps/fsaverage-sub12_XZUKLHYZ_AE-morph.fif`

* `$SUBJECTS_DIR/surf/lh.white` and `rh.white`
* the `stc` (with `stc.save`)
* the `inv` (`mne.minimum_norm.write_inverse_operator`)

I think that should be enough for me to reproduce and look at what's going on.

romquentin, 2017-08-08:
You can find these files here:
https://www.dropbox.com/sh/1nburpswo5lqesf/AAC0T4QpoFaK-BlYTlZS6-aka?dl=0
Let me know if you need something else. Thanks a lot!

Eric89GXL, 2017-08-09:
There is a mismatch between your source space surfaces and the Freesurfer surfaces that you have uploaded for your subject. When I run this:
```
import numpy as np
import mne

subjects_dir = '.'
subject = 'sub12_XZUKLHYZ_AE'
stc = mne.read_source_estimate('stc')
inv = mne.minimum_norm.read_inverse_operator(subject + '-inv.fif')
for s, hemi, vertno in zip(inv['src'], ('lh', 'rh'), stc.vertices):
    assert s['subject_his_id'] == subject
    rr, tris = mne.read_surface('%s/%s/surf/%s.white'
                                % (subjects_dir, subject, hemi))
    assert np.array_equal(s['vertno'], vertno)
    assert rr.shape == s['rr'].shape, (rr.shape, s['rr'].shape, hemi)```
```
I get:
```
AssertionError: ((161335, 3), (156287, 3), 'lh')
```
Is there somehow a problem in your pipeline with multiple `subjects_dir`s being used?

If I add this line to my script:
```
src = mne.setup_source_space(subject, subjects_dir=subjects_dir)
```
I get an error about `lh.sphere` missing (can you upload `lh.sphere` and `rh.sphere`, too?), but you should be able to provide the files yourself to try it. You should see that this `src` does not match `inv['src']`, even though it should -- so I think that somehow the wrong files have been used at some point.

romquentin, 2017-08-16:
It is working now. Thanks a lot!

closed 2017-08-16 by romquentin