mne-tools/mne-python #4471: Reading coordinates (montage) from vhdr files

jdue, 2017-08-09:
I have encountered an issue trying to read coordinate information (i.e., montage) from brainvision vhdr files (from the 'coordinates' field). For example, something like this

    raw = mne.io.read_raw_brainvision('filename.vhdr', montage=True, eog=['EOG1', 'EOG2'])

results in `ImportError: cannot import name '_sphere_to_cartesian'` as this function does not seem to exist in mne.transforms. Another function, `_sph_to_cart`, does exist, however. The expected argument structure of this function differ slightly though.

One way to solve this issue might be to modify line 438 of mne.io.brainvision from

    from ...transforms import _sphere_to_cartesian

to

    from ...transforms import _sph_to_cart

and, consequently, also changing line 446 from

    pos = _sphere_to_cartesian(r=radius, theta=theta, phi=phi)

to 

    pos = _sph_to_cart(np.atleast_2d([radius, theta, phi])).squeeze()

to conform with the expected structure of arguments of `_sph_to_cart`. This seems to solve the issue for me at least but I don't know what your preferred solution is, e.g., if you would rather modify `_sph_to_cart`.

I am using MNE 0.14.1.

agramfort, 2017-08-09:
please check if it's still broken in master

remove your current mne version with pip uninstall mne
then:

git clone git@github.com:mne-tools/mne-python.git
cd mne-python
pip install -e .

you should be able to modify the mne-python and send us PRs.

see also:

http://martinos.org/mne/dev/contributing.html

larsoner, 2019-04-16:
Fixed by  #4531

closed 2019-04-16 by larsoner