mne-tools/mne-python #4474: Doubts!! 'mne.stats.spatio_temporal_cluster_1samp_test' function documentation Error?

Diptyajitd7, 2017-08-09:
I am doing non-parametric cluster-level 1 sample testing. I am using the standard function from mne, named 'spatio_temporal_cluster_1samp_test' function for extracting information from source level MEG data.

While using this function from mne package, I am facing some doubts. As per documentation in the mne, this  function will return H0 parameter which is  the maximum cluster level statistics under permutation. H0 should have same array of shape with n_permutations(no. of permutation defined in the function as input).

Here comes the trouble, with the data I have, my H0 parameter doesn't have the same shape of n_permutations. H0 size is always less the n_permutations.

I also tried out the default mne's example : Permutation t-test on source data with spatio-temporal clustering test. (link:https://martinos.org/mne/stable/auto_tutorials/plot_stats_cluster_spatio_temporal.html)

Here too, I have seen the sample effect that H0 size is not same with size of n_permutations.

Opinion:

H0 size is not same as n_permutations number, Is it because some the maximum clusters sum (statistical weight) are repeated while shuffling process and function doesn't count the repetitive ones ? Correct me if I am wrong. If this is the case then mne documentation is little misleading I feel. It should be corrected.

Or, Did you use some internal threshold to omit some of the small weighted clusters from maximum cluster level statistics(H0) to make the calculation faster?

It would be nice if you can clarify on this issue. Because, I feel the threshold value and size of H0 play a crucial role in terms  of calculating p_values for all initial clusters.

Kind Regards,
Diptyajit Das

agramfort, 2017-08-09:
len(H0) should depend of n_samples as you cannot have for example 1000
permutations
if you only have 5 samples. Yet for K samples one should have 2**K
possible permutations
for a one sample test which is not what we have. You seem to get always less.

@Eric89GXL did you write this code?

Diptyajitd7, 2017-08-09:
Also interesting fact, after using this one sample test function, 
Even though you keep your sample no. fixed(say 10 samples), if you change the threshold value then len(H0) differs which is expected. If you provide higher threshold then cluster mass weight for H0 values becomes smaller and len(H0) increases. if You provide low threshold, cluster mass weight for H0 values increases and len(H0) decreases.So  Len(H0) becomes independent of len(n_permutations). This is my experience with MEG data that I am headlining. So it seems like len(H0) mainly depends on threshold value rather than on the permutation number and sample no. at lest for (one sample test function).

Eric89GXL, 2017-08-09:
The docs should probably say that the *maximum* shape is `(n_permutations,)`. If no clusters are found in some of the permutations, it can be smaller.

Eric89GXL, 2017-08-09:
@Diptyajitd7 can you open a PR to improve the doc?

Diptyajitd7, 2017-08-09:
@Eric89GXL I have created a PR. 
Can you please en-light the fact about 'no clusters are found under some of the permutations' for me?   
thanks!

Eric89GXL, 2017-08-09:
Actually I realized that when no clusters are found, it should still contribute to H0 -- it should contribute a cluster of size 0. I'll make a fix for it.

Diptyajitd7, 2017-08-09:
That actually makes a lot more sense. As then it will also change the p_values of all the initial clusters.
Don't get me wrong, I still somehow don't get the idea of not finding any cluster under some permutations. How can it is possible to not find any clusters at some permutation when you are summing up the statistical values of adjacent time points after applying a certain threshold? I guess the cluster size can be really small but not equal to zero. Or is it, the cluster can't be simply formed with a single time point? then you have a cluster size of 0?

Eric89GXL, 2017-08-09:
> summing up the statistical values of adjacent time points after applying a certain threshold?

Other order -- you apply a threshold, then sum statistical values (that exceed that threshold). So if no pass the threshold, you can have no clusters.

Diptyajitd7, 2017-08-09:
I see..yes I agree with you. Thanks a lot for your time.

Eric89GXL, 2017-08-09:
Actually in testing I can only seem to get `H0.shape == (0,)` or `(n_permutations,)` do you have an example that creates something in between? Here is what I have been testing with:
```
rng = np.random.RandomState(0)
data = rng.rand(20, 100, 1) - 0.5
t, clust, p, h0 = mne.stats.spatio_temporal_cluster_1samp_test(
        data, threshold=100, n_permutations=1024, seed=0)
```

Diptyajitd7, 2017-08-09:
did you try to run the example on the mne? here is the link:
 https://martinos.org/mne/stable/auto_tutorials/plot_stats_cluster_spatio_temporal.html 
if you use this example and meg sample data,  you should get H0.shape  is equal to 63, it doesn't matter if you use n_permutation as 1024 (default) or 2000 or 5000. H0 size remains same.

In case if you want some other data for testing, I am willing to share of my experimental data with you. But for that I have to upload the data to a server. Let me know you want that.

Eric89GXL, 2017-08-09:
Ahh okay, it's a problem with the 2-sample tests -- I can confirm that and will look. 

Your initial report was about the 1-sample tests, and there I only ever see `0` or `n_permutations` lengths --
 can you confirm (or provide a counterexample)?

Diptyajitd7, 2017-08-09:
I will try to put an example soon.

Eric89GXL, 2017-08-09:
Oh actually that is a 1-sample test. 63 is the correct number there -- there are only 7 "subjects", so it performs an exact test with `2 ** 6 - 1 == 63` flips.

Diptyajitd7, 2017-08-09:
so H0 size depends on the sample numbers((2**subjects -1) in this case?) not only the n_permutation size that you provide? Wait i am going to put up some counter example then it will be more clear to you

Eric89GXL, 2017-08-09:
> so H0 size depends on the sample numbers((2**subjects -1) in this case?) not only the n_permutation size that you provide

In the 1-sample test case, yes, because we can do an exact test. Basically it's `min(n_permutations, 2 ** (n_subjects - 1) - 1`

Diptyajitd7, 2017-08-09:
I see..so for one sample t- test, if (2 ** n_subjects -1)< n_permutations, then H0.size == (2 ** n_subjects -1). This is valid because of the small sample size?

Diptyajitd7, 2017-08-09:
I mean, (2 ** n_subjects -1)-1< n_permutations, then H0.size == (2 ** n_subjects -1)-1

Eric89GXL, 2017-08-09:
It's valid in the sense that you have now done an exact test (done all possible sign-flips), so there is no reason to do any others.

Diptyajitd7, 2017-08-09:
Ok.. whats your take on 2-sample tests, as you mentioned in earlier comment that there could some issue. I didn't find anything though.

Eric89GXL, 2017-08-09:
Those appear to be fine. I'll open a PR to document this better, though.

Diptyajitd7, 2017-08-09:
Sounds great! Thanks a lot again. cheers!

Diptyajitd7, 2017-08-16:
@larsoner  for the 2-sample test case, the formula, min(n_permutations, 2 ** (n_subjects - 1) - 1) is also valid  or the function avoids the formula and only takes no. of n_permutations that you assign during the test. I am trying to understand the effect of different number of permutations to the test. I have data size (10, 203,20800) (subjects*times*vertices) in one test condition and in another test condition data size is (11, 203,20800). If I want to find a difference between these two test conditions using two sample (F test) , the maximum no. of permutation that can be done is (2**n_subjects -1)? in this case (2**21 -1) equals to 2097151? Whats you take on the effect of doing less permutation to two sample test?

larsoner, 2017-08-16:
The 2-sample case does not use this formula and runs whatever you give it for `n_permutations`

Diptyajitd7, 2017-08-16:
I see.. but is there any rule of thumb or approximation that can be done to at-least have an idea about how many permutation is suitable for your data especially two sample test? Can you do an approximation just looking on the no. of observations/subjects you have in your data? If we can't have an approximation about the n_permutations that is suitable for the data then somehow it limits the model predictions. One has to emphatically choose/change (i.e. 10k, 20k, or even 50k) to different permutation numbers to get some results every time. One will be always unsure about the results as permutation no. can be reach up to infinity that may change the results to some extent.

larsoner, 2017-08-16:
> One has to emphatically choose/change (i.e. 10k, 20k, or even 50k) to different permutation numbers to get some results every time. One will be always unsure about the results as permutation no. can be reach up to infinity that may change the results to some extent.

These should have asymptotic behavior, so something like 1000 is probably not enough but something like 20000 probably should be. You can try gradually increasing the number, and the values shouldn't change much.

But this question is better asked on the listserv, because it's less about code / bugs / enhancements and more about how methods work / how to apply them.

Diptyajitd7, 2017-08-25:
@larsoner Hii there! So far you have been a great help for me to understand the Spatio-temporal clustering method better. I would really appreciate if you can manage some time and able to clarify my a doubt of mine.

**Doubt: < _Source space Data_ >** 
Situation: I am currently analysing MEG data recorded from 3 different groups of subjects. Each group of subject has 4 experimental conditions. (2 congruent and 2 in-congruent conditions.)  
For example: 
2 congruent conditions are : LLst and RRst
2 in-congruent conditions are : LRst and RLst

I am interested in finding differences in  (LLst Vs LRst) (congruent Vs in-congruent) conditions within a _specific group of subjects_. p.s: two conditions are independent from each other but belongs to same group of subjects. Now I want to use spatio-temporal clustering.

Possible Solution: 
Do a **One Sample spatio-temporal clustering test** where test statistic is **a paired t -test (Mean difference test)** (paired contrast between LLst Vs LRst ).  

Drawback: When the brain signals are recoded for each condition, the synchronization of the signals during recording may differ for two test conditions. Therefore, time sample and Voxel/vertices sample may shift in case of two test conditions. Therefore, paired contrast between LLst Vs LRst may not be ideal or sufficient enough to draw a conclusion that those two test conditions differ from each other.

Doubts I have: 
Is there somehow, I can utilize  **Two samples spatio-temporal clustering**  to find a difference between ( LLst Vs LRst ) test conditions from the **same group of subjects**? 

Note: Please correct me if I am wrong, the test statistic used in  **Two samples spatio-temporal clustering**  is a F-test (i.e., Variance test). The main assumption of a F- statistic is, the variance testing should be done on the data coming from _two different groups of subjects_.  This assumption is violated when I want to do a test ( LLst Vs LRst ) from a same group of subjects although my two test conditions are independent from each other. So is it wise to use  **Two samples spatio-temporal clustering**  in this case? 

I have doubt on my assumption that Two samples spatio-temporal clustering test may not be appropriate to handle kind of problem (LLst Vs LRst: two conditions from same group of subjects). I feel **Two samples spatio-temporal clustering** are specific test that allows to find answers  when I want to test two conditions from _two different groups of subjects_  [for example: LLst (group 1) Vs LLst (group2)].

As the test is delicate, your help or suggestion will be really helpful. In case if you are not sure then kindly suggest someone to whom I should contact.

Kind regards!

larsoner, 2017-08-25:
This question is not really related to code problems (what the GitHub tracker is for) and I'm not sure I'll have time to look and answer -- could you please post to the MNE listserv instead? That way the community can weigh in

PR #4478 (2017-08-29): MRG+1: Better rng handling in cluster functions
Closes #4474.
closed 2017-08-25 by larsoner