mne-tools/mne-python #4480: Current Source Density (CSD) with mne-python?

giuliagennari, 2017-08-10:
I would like to transform my EEG data from voltage into current density. 
Something like this http://psychophysiology.cpmc.columbia.edu/Software/CSDtoolbox/tutorial.html 
but through Python, can anyone give me some instructions?

I am a young PhD student, this would be extremely helpful!!!

Thanks a lot in advance!

larsoner, 2017-08-11:
We currently do not have CSD estimation in MNE, but it is something we plan to implement. 

@dengemann do you have time to give it a try?

jona-sassenhagen, 2017-08-30:
Denis is married now, he's done writing code :D

dengemann, 2017-08-30:
Epic provocations to get me coding here! You guys are married too :) My MNE
coding priority is in auroreject these days and some maintenance issues.
But happy to supervise / help with things.

mmagnuski, 2017-08-30:
I will need CSD in a month or so - I could try this in a few weeks, but I would need help.

mmagnuski, 2017-08-30:
(BTW I am not married so I can still code ;) )

agramfort, 2017-08-30:
imagine the day you have kids :)

palday, 2018-03-13:
I took a look/thought about this a bit .... 

There are some subtle data structure issues with CSD data -- [they're lot like normal EEG data, except when they're not](http://psychophysiology.cpmc.columbia.edu/Software/CSDtoolbox/faq.html#PostProcessing) ([different units, reference-free](http://psychophysiology.cpmc.columbia.edu/Software/CSDtoolbox/errors.html#CSDReference), [lack of volume conduction](http://psychophysiology.cpmc.columbia.edu/Software/CSDtoolbox/faq.html#CSDinverse), etc.). This leads to the obvious question of what channel type to assign them. If we let users be responsible for not doing invalid transformations on CSD data, then `eeg` seems like generally the obvious choice with the exception of the units, which becomes relevant for plotting. It would also be nice to somehow indicate that it's CSD data when saving and loading ([it's not harmless to re-apply the CSD transformation](http://psychophysiology.cpmc.columbia.edu/Software/CSDtoolbox/errors.html#CSD2CSD) unlike re-referencing, which is normally obvious anyway), but the FIFF format doesn't have a CSD channel type. the closest thing seems to be the `ch_unit` tag, but that's an `enum`, which doens't include `uV/cm^2` as a possibility. Another option may be to include the CSD as a projection, but again, the corresponding `enum` doesn't have a value for it. Yet another alternative would be to treat CSD like we currently treat ICA, but the big difference there seems to be `ICA.apply()` yields "normal" EEG data back. 

Otherwise, I think a lot of the more numerical steps for CSD could be achieved by re-factoring some existing code, e.g. parts of channel interpolation uses some of the same techniques as used in the usual CSD calculations.

jona-sassenhagen, 2018-03-13:
I think given that we don't take care of  any of these things for, e.g., the Evoked objects containing log p values, or ICA-transformed objects, I think it'd be consistent to just implement CSD for now, leave it to the user to not fuck up things here, and think about stronger support later.

@miykael

larsoner, 2018-03-13:
As far as dealing with the FIF format goes, we can probably get away with storing the CSD transformation in `raw.info['comps']`. Usually this applies only to MEG, but IIUC we can store any linear transformation in there, and it will be handled properly in forward calculations, etc.

jona-sassenhagen, 2018-03-13:
I like that option.

transpersonify, 2018-09-11:
Is there any update on this issue?
I am looking for a python version of CSD to apply on my EEG data (I have channel x time data matrix and 2D location matrix, independent of mne layout). I found implementation suitable for LFP but not for EEG.  Can anyone help?

larsoner, 2018-09-11:
It still has not been implemented AFAIK

sherdim, 2018-11-14:
Look at [adaptaton of CSDtoolbox functions](https://gist.github.com/sherdim/28b069c8ebca17073f24322e8e721e1f)

It will be nice to get them somewhere in  `mne.decoding.transformer`

mmagnuski, 2018-11-14:
Good you're sharing this, I had my own adaptation of CSD tested against CSD toolbox for a some time now (https://github.com/mmagnuski/borsar/blob/master/borsar/csd.py) but I didn't have time to make a proper PR. Do you think you could do it? Because this is an adaptation we would also need to ask the original authors about licensing. I can help with testing etc. (I have some tests for this already).

larsoner, 2018-11-14:
IIRC @dengemann has already talked to the authors of the CSD MATLAB toolbox and obtained permission to adapt to MNE-Python and re-license under BSD. @dengemann can you confirm?

jona-sassenhagen, 2018-11-14:
Would be cool.

sherdim, 2018-11-14:
I am ready to participate but I dont have idea about `raw.info['comps']` mechanism supposed by @larsoner .  Also routines in _get_GH could be optimized more I think (in MATLAB even np.eye was done through cycle). And yes, need help with testing.

mmagnuski, 2018-11-14:
Yes, matlab does some JIT compilation IIRC so too many loops may not be such a problem as in python. But in general it is better to have it working than not have it all. And optimize once it is in place and works. I'd first create array functions for csd, as you did, then add interface for raw files and worry about storing the transformations in `['comps']` later. And you already have a skeleton for a tutorial in the shared notebook.
As for tests: I already have a simple test against results computed with CSD toolbox here: https://github.com/mmagnuski/borsar/blob/master/borsar/tests/test_csd.py, we may start with that.

larsoner, 2018-11-14:
Yes don't worry about `comps` for now. Or efficiency. These can be optimized.

Let's get it working with an example we can all cross-check and play with. Then we can get others on the more annoying/fun aspects, depending on your thoughts about optimization and API design :)

FWIW I think some or most of the code necessary to do this already lives in private functions used for EEG interpolation (which is probably already decently optimized). @dengemann might also have some separate private code. We should take the best of each.

dengemann, 2018-11-15:
Take a look here, it's not private:
https://github.com/nice-tools/pycsd?files=1 (currently GPL3-licensed).
Jürgen Kayser agreed that we can relicense this code and move / translate
into MNE.

Denis

agramfort, 2018-11-15:
we just need a volunteer to do it :)

closed 2019-11-13 by agramfort