mne-tools/mne-python #4481: Collaboration with CEBL3 project?

brylie, 2017-08-11:
CEBL3 is a BCI platform built at Colorado State University Fort Collins BCI Lab:

https://github.com/idfah/cebl/

What would be some possibilities or considerations regarding collaboration between the MNE-Python and CEBL3 development communities?

Related issue:

agramfort, 2017-08-11:
@brylie sure but what do you have in mind? do you see some duplication of efforts ? some possible synergies? if so on what?

brylie, 2017-08-11:
Well, the first common thread is Python. Basically, I am hoping that CEBL3 and MNE-Python might be comparable in scope to BCILAB or OpenViBE. It just seems like a good connection to make.

agramfort, 2017-08-12:
sure why not. maybe @alexandrebarachant or @cbrnr who are MNE devs + BCI experts that give you recommendations.

brylie, 2017-08-14:
Thanks @agramfort for your quick responses, and for further inviting people to the discussion.

@idfah what are your thoughts about MNE-Python as it might relate to CEBL3?

cbrnr, 2017-08-16:
Without having looked at the CEBL3 code base in detail, I'd say that there is potential overlap in the offline analyses area. That is, CEBL3 could benefit from existing MNE functionality for example by using our filter implementations, PSD, various topomap visualizations, montage workflow, and so on. I'm currently working on a GUI for MNE (https://github.com/cbrnr/mnelab) that will (hopefully) make data analysis with MNE more accessible for beginners. The thing that CEBL3 adds on top of all that is online capability. So I don't know if this is worth the effort (certainly several data structures would need to be adapted), but I think it would be great if you re-used as much existing stuff as possible (which means that MNE stuff gets more testing too).

On a related note, you should consider adding [LSL](https://github.com/sccn/labstreaminglayer) support as the BCI community seems to gravitate towards this protocol.

brylie, 2017-08-16:
Thanks for the response @cbrnr. I have also emailed Dr. Chuck Anderson from the CSU BCI Lab regarding  possible collaboration.

The MNE-Lab GUI looks nice, and I appreciate that it is built with Qt.

The proposal for LSL support in CEBL3 also seems like a good direction.

@idfah what are your thoughts on MNE-Lab and CEBL3 overlap/collaboration, and/or the possibility of CEBL3 supporting the [Lab Streaming Layer](https://github.com/sccn/labstreaminglayer)?

alexandrebarachant, 2017-08-16:
I'm joining @cbrnr on his comment. There seems to have a large overlap on offline analysis, and MNE looks more mature in this area. But the real-time module can be a good convergence point.

I also strongly encourage the use of LSL, as it will provide you support for most of the EEG device on the market.

jona-sassenhagen, 2017-08-30:
ping @teonbrooks

idfah, 2017-09-04:
Glad to see there is interest in potential collaboration between the MNE and CEBL projects!  I definitely see some overlap and also think that each project has some functionality that is not found in the other.  MNE appears to me to be more mature in many ways, but I think CEBL has some machine learning and signal processing functionality that isn't currently found anywhere else.  The CEBL gui also has some nice functionality that is very oriented toward BCIs.  I would be willing to look at LSL, but migrating that way would likely be down the road a bit.

At the moment, I am in the throws of completing my PhD dissertation and transitioning to a new job.  Once things settle down a bit I would be happy to take a closer look at potential avenues for collaboration.  May we keep this issue going for a while?

larsoner, 2017-09-04:
> CEBL has some machine learning and signal processing functionality that
isn't currently found anywhere else.

Which processing / ML bits in particular are we missing at the moment?

> May we keep this issue going for a while?

Sure, we'll probably periodically bother you about it as a reminder, though
:)

idfah, 2017-09-04:
At a glance, it appears that MNE has lots of the same things I have implemented in CEBL, including PSDs, CWTs, AR models, FIR and IIR filters, interpolation and resampling.  MNE generally has a lot more options and much better documentation, but CEBL generally follows an object oriented design that may be easier to use.

> Which processing / ML bits in particular are we missing at the moment?

Maximum Signal Fraction (MSF) and Principal Components Analysis (PCA) come to mind.  I think I saw ICA and CSP in MNE, they are also in CEBL.  CEBL also allows these transforms to include time lags.  I really don't see much ML stuff in MNE, I might be missing it or else MNE may assume the use of other ML packages.  Some of the algorithms in CEBL are designed specifically for EEG signals and BCIs, i.e., conv nets and recurrent nets.  We also have really good neural net, discriminant analysis and optimization algorithms.

Many of the features in CEBL have come directly from the needs of my research.  So, it also includes various algorithms that I have been investigating.  Some published and some not.

larsoner, 2017-09-05:
> CEBL generally follows an object oriented design that may be easier to use.

The biggest place I think this could be useful is in filtering, where you might want to operate on segments at separate times (e.g., as epochs come in) as if they were continuous, and thus you need to retain and reuse state information, which could be useful for BCIs. We could do this in MNE if it makes sense to unify at some point.

BTW when I briefly looked at the code in your repo I noticed that your resampling code path uses upsample->FIR->decimate -- you might want to switch to the `resample_poly` in SciPy because it gives nice speed gains (using polyphase filtering to avoid the unnecessary computations of output samples that get discarded during the decimation step anyway).

> Maximum Signal Fraction (MSF) and Principal Components Analysis (PCA) come to mind... I really don't see much ML stuff in MNE, I might be missing it or else MNE may assume the use of other ML packages. Some of the algorithms in CEBL are designed specifically for EEG signals and BCIs, i.e., conv nets and recurrent nets. We also have really good neural net, discriminant analysis and optimization algorithms.

Yes mostly we use `scikit-learn` for ML purposes, with wrappers and extensions in the [decoding module](https://mne-tools.github.io/dev/python_reference.html#decoding). For optimization problems we mostly rely on `scipy.optimize`.

agramfort, 2017-09-05:
I am curious about the recurrent nets and optimization algorithms.

can you tell us more?

kingjr, 2017-09-05:
> CEBL also allows these transforms to include time lags. 

You mean build and Hankel matrix? We can add this transformer if there is a clear use case.

jona-sassenhagen, 2017-09-05:
>You mean build an Hankel matrix? We can add this transformer if there is a clear use case.

I'd like that ... there should be a basis [here](https://github.com/mne-tools/mne-python/blob/master/mne/stats/regression.py#L319)

larsoner, 2017-09-05:
the encoding/STRF code also has a function to delay time samples

larsoner, 2024-09-19:
CEBL3 looks to be maybe not under such active development anymore (?) and I'm not sure we landed on any points of collaboration some 7 years ago :) But let's reopen if opportunities or new discussions arise!

closed 2024-09-19 by larsoner