mne-tools/mne-python #4484: ENH/VIZ: display .nave by default in plot_evoked

dengemann, 2017-08-14:
I was thinking repeatedly for months that having the .nave output by default in the title or so of plot_evoked would be so useful. What do you think?

cc @jona-sassenhagen @mmagnuski

larsoner, 2017-08-14:
A parenthetical after the condition name would be nice

jona-sassenhagen, 2017-08-14:
Probably better than what we have right now :)

dengemann, 2017-08-14:
> A parenthetical after the condition name would be nice

yes

> Probably better than what we have right now :)

and yes.

Great, so it seems some people like it. Let's do it! Marking as easy.

dengemann, 2017-08-14:
@massich maybe something you could like to look into, relaxing from pytests.

mmagnuski, 2017-08-14:
I like that too :)

dengemann, 2017-08-14:
> I like that too :)

great!

thesujitroy, 2018-02-25:
Hi,
Can I get some reference to the problem and what sort of script is required to be written.
And this is related to which objective.

thesujitroy, 2018-02-25:
Or if I can get some doc to look into

dengemann, 2018-02-25:
Hi @thesujitroy have a look at some of the basic tutorials that compute evoked responses and call ```evoked.plot()```. You will notices that no information on the number of averages (nave) is present. The idea is to find a way to show the evoked.nave attribute when plotting. mne.viz.plot_evoked needs to be updated for that. I hope this helps.

thesujitroy, 2018-02-25:
Alright. Thank you

thesujitroy, 2018-03-02:
I  am taking this issue.

dengemann, 2018-03-02:
Great! Don't hesitate moving in small steps. It is bettet to start
somewhere and communicate early. Perfection comes in iterations.

thesujitroy, 2018-03-08:
Hi,
I was going through the code and it seems that nave is declared to be 1.
However, in plot_evoked the attribute needs to be added. But I cant understand the attributes we are trying to show is evoked.nave, this needs to show in plot in what form.
Can the Number of averages be more than 1, and are trying to put it in butterfly plot?

jasmainak, 2018-03-10:
> But I cant understand the attributes we are trying to show is evoked.nave, this needs to show in plot in what form.

See comment [from above](https://github.com/mne-tools/mne-python/issues/4484#issuecomment-322205698)

> Can the Number of averages be more than 1, and are trying to put it in butterfly plot?

It's the number of trials used in the average for the evoked, i.e., evoked.nave

jona-sassenhagen, 2018-03-31:
@thesujitroy any progress?

PR #5100 (2018-04-05): MRG: Append nave to evoked.plot
Closes #4484.

@jona-sassenhagen I don't think it looks so pretty:

[image]

How should I make it nicer?
closed 2018-04-05 by jona-sassenhagen