mne-tools/mne-python #4486: DOC: broken link for [source]

aweinstein, 2017-08-14:
In the documentation of the API, the link to the source, that appears as "[source]" is broken.

For instance, the documentation page for [read_montage](https://martinos.org/mne/stable/generated/mne.channels.read_montage.html#mne.channels.read_montage) has this [source](http://github.com/mne-tools/mne-python/blob/maint/0.14.1/mne/io/edf/edf.py#L552-L608) link, wich produces a GitHub 404 error.

larsoner, 2017-08-14:
Indeed, I've opened two PRs (one to master, one to maint/0.14) that should fix it

#4487
#4488

larsoner, 2017-08-14:
Thanks for the report @aweinstein the bug has been fixed, it should show up in the next ~hours or day when the automated build updates everything

closed 2017-08-14 by larsoner